Ultimaker/Cura #12513: Glitchy Pop-Up Windows

jk101290, 2022-06-13:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

Open any pop-up window such as preferences > configure cura

### Actual results

There are errors in the way the window is displayed 

### Expected results

The windows are displayed normally.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm using a laptop with 2 additional monitors attached. One 20 inch and one 22 inch at 1600 x 900 and 1920 x 1080. I see the glitch happen on both monitors but not on the 12 inch laptop screen which is at 1920 x 1080.

See pics below. When I move the window by clicking the top "bar" and dragging it, the issue corrects itself. But still annoying. This issue also occurs on the default theme.

[image]

[image]

nallath, 2022-06-15:
I will discuss the issue with the rest of the team!

Joeydelarago, 2022-06-28:
Closing this as a duplicate of [#12602](https://github.com/Ultimaker/Cura/issues/12602)