Ultimaker/Cura #12518: Cura does not auto reconnect to cloud (digitalfactory) printer after suspend

matthiasamberg, 2022-06-14:
### Application Version

5.0.0

### Platform

MacOS 12.4 Apple M1 

### Printer

Ultimaker S3

### Reproduction steps

**Before**:
* Have a digitalfactory.ultimaker.com account
* Have a Ultimaker Printer setup and connected to the digitalfactory cloud
* Have cure setup and logged in to the digitalfactory cloud so that you can see the connected printers

**To reproduce**
- Start Cura
- The cloud printer is shown as connected in the Prepare-View in Cura
- Close the computer lid (induce suspend/sleep). 
- Wait 1 Minute
- Wake computer up from suspend (open lid)
- The printer is now disconnected (the cloud icon is greyed out)
- The printer will not reconnect on it's own

### Actual results

The digital cloud printer is not connected to Cura anymore after suspend. It does not seem to reconnect either by waiting or any other activity (clicking on it, switching to "manage printers" or "add printers" screens.

### Expected results

A previously connected cloud printer should reconnect automatically after a computer has been put into suspend and then wakes up.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

there is no additional information or file uploads, why is this field  required?

matthiasamberg, 2026-08-16:
It's been years.. I've abandoned Cura and mostly Ultimaker for good, (and good riddance for that)... you gotta figure it out yourself man if it is still happening.. if response time is 4 years do you expect me to still respond?

GregValiant, 2026-08-16:
I'm trying to get a handle on all this crap.  I'll close this.

closed 2026-08-16 by GregValiant