Ultimaker/Cura #12520: Search and replace

ialsaheb, 2022-06-14:
### Application Version

5

### Platform

Windows 10

### Printer

ender 3 pro

### Reproduction steps

1. I add a script - search and replace.
2. i copied a paragraph.
3. in the replace area i can't past

### Actual results

you can't past a paragraph in replace area

### Expected results

in the earlier versions of cura you can past a paragraph in the replace area.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional information

nallath, 2022-06-16:
Great find @GregValiant! The multi-line is indeed the issue here. I couldn't reproduce it myself, but i was just copying single lines into it.

I will take it up with the rest of the team.

MariMakes, 2022-08-23:
Hey @ialsaheb 

Thanks for reporting! I had the same reason for missing it as @nallath since I copy pasted it line by line, great find that it's related to multi line @GregValiant 

We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.  

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker. We believe we will not work on this anytime soon and will therefore defer this issue. Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I hope you understand.

closed 2022-08-23 by MariMakes