Ultimaker/Cura #12523: Seam is always not aligned in curved objects

JimmyThePiper, 2022-06-14:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

irreleant

### Reproduction steps

Take any cylinder (for example), regardless of the seam alignment selected, its never in the same place along the layers; Cura 4.13 wasn't perfect either, but it was much better at making a fairly straight line. 5.0.0 sometimes even splits the seam in two lines!

### Actual results

A very chaotic seam line

### Expected results

a neat line

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

JimmyThePiper, 2022-06-14:
There should be a "seam corner preference" that's just "straight line", regardless of corners or curves

smartavionics, 2022-06-15:
See https://github.com/smartavionics/Cura/releases for a Cura variant that supports this concept:

With these releases, the combination of z-seam alignment = user specified and seam corner preference = none you get this:

[image]

Rather than this:

[image]