Ultimaker/Cura #12524: World space to local Space

petermt08, 2022-06-15:
### Application Version

5.0

### Platform

windows 11

### Printer

ultimaker 5

### Reproduction steps

I go to scale my model and it works fine and then I and my support blockers and I go to and duplicate my model and group and then go to rotate is on the x axis and my y and z axis swap dimensions with each other so if y was 10 and z was 9 y then become 9 and z become come and no matter which axis you chose to rotate it just changes the axis's of the two that u dont rotate off I have tried everything to make this work , I the to scale it after its roates but then the model becomes all warped looking.

### Actual results

So I know alot of you are going to say this is a duplicate but it is not I have been on the phone with bothe cura and microsoft both say there software is fine but yet something is wrong. I have reach out and have gotten feedback from greg i have tried his methods none of them have worked here is the kick I installed a few more 3d programs on my windows 11 and they are all doing the same thing please someone help me out the issue is as follows I go to scale my model and it works fine and then I and my support blockers and I go to and duplicate my model and group and then go to rotate is on the x axis and my y and z axis swap dimensions with each other so if y was 10 and z was 9 y then become 9 and z become come and no matter which axis you chose to rotate it just changes the axis's of the two that u dont rotate off I have tried everything to make this work , I the to scale it after its roates but then the model becomes all warped looking. the funny thing is the in other 3d programs when you go to rotate your model the model does not rotate in place the actually model moves to the degree you chose so if you rotate on x axis 90 the actual model moves up 90 degrees as well and doesnt rotate in place.

### Expected results

The model should rotate and scale properly with no issue

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[output_log.txt](https://github.com/Ultimaker/Cura/files/8904710/output_log.txt)
 or ways to do so

rburema, 2022-06-15:
Hi! It was somewhat difficult to parse your sentences, since you don't use punctuation, or screenshots. My interpretation of what you actually meant might be wrong as a result.

What I think is going on is expected behaviour: The measurements you see are for the bounding-box of the object in world-space. We do not show the local-space measurements of the object anywhere in the interface.

If you need to adjust your model precisely in its own local space, please use a CAD/CAM program.

MariMakes, 2022-08-23:
Hey @petermt08,

just as @rburema I'm struggling with troubleshooting your bug with so little information. 

Can you provide for me the following sentence:
**I see ... but I expected ...**
+ a .3mf file which you are running into trouble with? 
You can create a zip from it to attach it to this github ticket. 

If your issue is already resolved, do you mind closing this issue for me? 

Thank you!

closed 2022-09-10 by github-actions[bot]