Ultimaker/Cura #12525: Z seam alignment when using Fuzzy Skin not on Corner

Teacher99, 2022-06-15:
### Application Version

5.0

### Platform

Windows 10

### Printer

Ender 5 Pro and Taz 6

### Reproduction steps

1. Use any square shape such as calibration plug in cube
2. Set fuzzy skin on and Z seam will jump from corner to down the wall a short distance and then refuse to switch corners even if you change seam preference. Rotating the model will it to move to a different corner relative to the bed.

### Actual results

Before turning on fuzzy skin, z seam will be on corner.  Turn on fuzzy and it jumps to down the wall a short distance, this was not happening in earlier Cura version.  If I select z seam preference location it is also ignored when using fuzzy skin.

### Expected results

Z seam preference and locations match with or without fuzzy skin.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

smartavionics, 2022-06-15:
Try using the user defined alignment with coordinates like 0, -100

Teacher99, 2022-06-16:
When I tried using coordinates no matter what I put in the Z seam never changes.  I actually discovered this problem when I was trying to get coasting to work as I was watching coasting seams in the preview and finding coasting doesnt work with my equipment on 5 and then noticed I could not change the Z seam location to hide it better on real prints.

PhilBaz, 2022-07-09:
Yes, just found the same issue....using fuzzy skin no seam options seem to make any impact, nether location or seam corner preference.

elsitiodelavilla, 2022-07-29:
I experience the same problem I have been printing the same piece for several months with fuzzy skin and random z seams without problems, Cura 5 alters the position of the seams and also once altered it is not possible to modify it, it always stays in the same place even changing settings

MariMakes, 2022-08-23:
Hey @Teacher99,

Thanks for your report, we've seen a number some similar reports.
This seems to be the same issue as  https://github.com/Ultimaker/Cura/issues/12190.

That issue is on our backlog with the intent to resolve it for the next release. 
I'm closing this issue, since it's a duplicate you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12190

Fingers crossed 🤞

closed 2022-08-23 by MariMakes

nallath, 2022-09-08:
This has been fixed for 5.2

PhilBaz, 2022-09-08:
@nallath any chance you can give us a hint of when the might be coming out?

elsitiodelavilla, 2022-09-09:
Very happy to hear this news. I'm still using version 4.13.1 just because of this problem