Ultimaker/Cura #12533: Cura 5.0 Print Settings List drop down - Cannot Extend

TimothyXYZen, 2022-06-15:
### Application Version

5.0.0

### Platform

Windows 11 Pro

### Printer

CR10S Pro V2

### Reproduction steps

1. I Installed Cura 5.0.0 yesterday and carried over my print profiles as usual when updating Cura.  
2. I uninstalled my old Cura 4.13.0
3. I Opened Cura 5.0.0 and the Print Settings list menu is locked to only shown 2 or 3 settings at a time.

[image]

### Actual results

I am unable to click & drag the bottom bar on the print settings drop down to extend it.  It is locked in length. See where I circled in Red.  I cannot drag this bar downwards to expose more print settings.

[image]

### Expected results

I should be able to click & drag the list down per normal Cura software functionality.  If I'm mistaken, please correct me in how to extend the list
[CCR10SPRO_ElectricalGuideRail-TimothyXYZen.zip](https://github.com/Ultimaker/Cura/files/8911402/CCR10SPRO_ElectricalGuideRail-TimothyXYZen.zip)
.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

No log file provided, let me know if it's necessary, but I have no crashes occuring.

nallath, 2022-06-16:
If you hover over the the dots in the bar, does your mouse cursor change? Does it matter if you have it on your main screen or on an external / extra monitor?

TimothyXYZen, 2022-06-16:
Yes, the two vertical arrows (one up and one down) appear but no matter how I click and drag it won't budge.No, I just tried main screen vs second monitor and it didn't make a difference.  I tried restarting my PC with my second monitor disconnected, no change.

TimothyXYZen, 2022-06-20:
This issue still persists.