Ultimaker/Cura #12535: Missing infill

paulnaude, 2022-06-15:
### Application Version

5.0.0

### Platform

Win 10

### Printer

Wanhao i3 Mini

### Reproduction steps

Slicing parts produces no infill, except for the first layer.

### Actual results

No infill, except for the first layer

### Expected results

Infill should be generated

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Only the first layer of the infill is created, the rest are not.

[image]
[WDI3M_Knob_1.zip](https://github.com/Ultimaker/Cura/files/8911812/WDI3M_Knob_1.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8911825/cura.log)

GregValiant, 2022-06-15:
Hello.
Changing the infill from "Concentric" to anything else causes the problem to go away.  Lowering the Infill Density to 25% and with Concentric makes it looks like this.
[image]

Which isn't right.  Someone from the Cura Team will take a look.
Was this something you just loaded into 5.0 or was it a "project file" created in a previous version and then opened in 5.0?

GregValiant, 2022-06-15:
Thanks.  There have been problems reported when opening a project in 5.0 that was created with an earlier version.  This appears to be different.

nallath, 2022-06-16:
It seems to be caused by the "Infill Layer Thickness" setting. If you set it to be the same as the layer height again, it does work.

I will discuss this issue with the rest of the team!

MariskaMaas, 2022-06-28:
Thanks for reporting and the extra investigation. It's been discussed with the team and is now a ticket on the backlog. 
For internal reference, it can be found under CURA-9426