Ultimaker/Cura #12537: Release Tags are not available via api.github.com

seemantk, 2022-06-15:
### Application Version

5.0.0

### Platform

all

### Printer

n/a

### Reproduction steps

Hello!  I am from Funtoo Linux and we use an automated system to make the latest upstream packages available to our users.  This automation uses the Github API to find the latest release.  It finds 5.0.0 mentioned at [https://api.github.com/repos/Ultimaker/Cura/tags]( at the repos/Ultimaker/Cura/releases API endpoint), but the [https://api.github.com/repos/Ultimaker/Cura/tags](tags endpoint) has very strange release numbers in it:

`'v4.5-beta', 'v4.4.1', 'v4.4-beta', 'v3.1.0-beta', 'v3.0.3', 'v2.1.3', 'lulzbot-21.04', 'lulzbot-21.03', 'lulzbot-21.02', 'lulzbot-21.01', 'lulzbot-21.00', 'lulzbot-20.03', 'lulzbot-20.02', 'lulzbot-20.01', 'lulzbot-19.15', 'lulzbot-19.14', 'lulzbot-19.13', 'lulzbot-19.12', 'lulzbot-19.08', 'lulzbot-19.08-1', 'lulzbot-19.07', 'lulzbot-19.05', 'lulzbot-19.04', 'lulzbot-19.03', 'lulzbot-19.02', 'lulzbot-19.01-1', 'lulzbot-18.05', 'lulzbot-18.05-1', 'lulzbot-18.04', 'lulzbot-18.03'`

As you can see 5.0.0 is not mentioned in the tags, so our automation fails to find the latest release.  Please note that this is also true of the CuraEngine repo

### Actual results

automation fails to find the latest release

### Expected results

automation makes the version 5.0.0 available to Funtoo Linux users

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

nallath, 2022-06-16:
Uuh, I don't really know what to do about that. The tags are definitely there, so it seems that the tags API from GH is either broken our outdated.

A workaround (one could argue that it's not even entirely a work around) is to use https://api.github.com/repos/Ultimaker/Cura/releases. The first result in that list is always the latest release that we did, and has a "tag_name" field that points to the correct tag. 

We don't have releases for the CuraEngine (or Uranium repo) but they do have the same tag_name. It would be a bit of extra work I guess, but I don't really see another way, as we can't control what the API of github does.

fieldOfView, 2022-06-16:
> The tags are definitely there, so it seems that the tags API from GH is either broken our outdated.

It looks like the github API wants the tags to start with a `v`, or something non-numeric and lowercase.

MariMakes, 2022-08-23:
Hey @seemantk,

Can you please let me know if this is still an issue for you? 👼 
Then I can bring it up to the team. 
They have been making changes to improve the build system. 

If the issue is resolved, could you please close the issue? 

Thank you! 💟

seemantk, 2022-08-24:
hi @MariMakes thank you.  Please allow me a few days, and I'll confirm and take the appropriate action here.  Thank you for being patient with my latency