Ultimaker/Cura #12540: Missing supports in higher layers, just before the first valid layer of the model.

hackingchips, 2022-06-16:
### Application Version

5.0.0

### Platform

Linux

### Printer

Anet A6

### Reproduction steps

1. Imported a STL model.
2. Sliced the model.
3. Printed the model and realized that the first valid layer of the model start printing in the air.
4. Previewed the model in Cura and realized supports are missing just after the first valid layer of the model.

### Actual results

At first i thought the problem was in the printer, but after previewing the model in Cura i realized supports where missing in some layers (higher layers of supports) just after the first valid layer of the model. So, in this zones, the printer start printing in the air.
I had this problem in Cura 4.11.0. Then i have installed version 5.0.0 and the problem persist. Behavior is very similar in Cura 4.11.0 and Cura 5.0.0.

Following animated gif shows the problem in Cura 4.11.0. The same problem happens using "normal" supports and using "tree" supports. Tested a lot of different settings (including layer height) and the problem persists.

[image]

Following animated gif shows the problem in Cura 5.0.0. If using normal supports, behavior is the same as in 4.11.0 version (missing layers). If using "tree" supports, supports are not created in some zones.

[image]

Following animated gif shows the problem in Cura 5.0.0 if orientation of the model is changed.

[image]

### Expected results

Supports have  to be printed up to the height of the first valid layer of the model so the printer does not start printing in the air.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The model.
[Left_With-Body.zip](https://github.com/Ultimaker/Cura/files/8917813/Left_With-Body.zip)

Asterchades, 2022-06-16:
The missing single and double layers immediately before the model are expected, and completely normal support behaviour across most, if not all slicers. The air gap or Z gap is to stop the supports and the model from fusing together, and should only be removed in the event you're using either soluble supports or support material which is incompatible with the main model material.

Should you want to try for yourself you can set the Support Top Distance to 0.0mm. By default this setting is 0.1mm for regular supports and 0.1mm plus one layer for Tree supports - setting it smaller makes the supports harder to remove but provide more support, while setting it larger makes the supports easier to remove but less effective. Note that Cura will always round the value up to the next layer when it comes to slicing.

The missing trees are peculiar, but unfortunately you didn't upload your project properly - it's completely blank. Chances are you didn't let it finish uploading before you submitted the post, so you'll need to upload it again.

hackingchips, 2022-06-17:
As you said @Asterchades, reducing the "Support Top Distance" to 0.0mm solves the problem.

In my first post, this value was configured at 0.1mm in all test except in version 5.0.0 "tree support" where it was configured at 0.4mm.
If i place same value in both versions, and all type of supports, behavior is the same. If i reduce "Support Top Distance" to 0.0mm, in both versions the supports are created up to the first valid layer.

So, as you said this is expected behavior and i took it as a bug because of my lack of knowledge about how supports are created.

if no one else has anything to say about this, i will close this issue in a few ours as it is not a bug.

closed 2022-06-18 by hackingchips