Ultimaker/Cura #12543: "slicing failed"

FjolnirsBath, 2022-06-16:
### Application Version

5.0.0

### Platform

windows 11

### Printer

Ender 3 Pro

### Reproduction steps

1. loaded file from stemfie rubber band car kit: SPS-000001 - Rubber-band-driven car - Beams Braces (stemfie.org) both by double clicking file in file explorer, and then by trying to open file from Cura file menu.
2. attempt to slice

### Actual results

receive unspecified "slicing failed" message, then forced to create this report

### Expected results

expected slicing to complete and file to export

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

inside text

nallath, 2022-06-17:
Please provide the model that fails for you. You can add them via the web interface if you zip them.

cht685, 2022-06-23:
Seems like this is becoming a popular error :( I just posted yet another one of these today. :(

closed 2022-07-01 by github-actions[bot]