Ultimaker/Cura #12548: Instruction link doesn't work

x40-Community, 2022-06-17:
### Application Version

5.0.0

### Platform

Ubuntu Linux 2022.04

### Printer

Ultimaker 

### Reproduction steps

I clicked on the book icon with the question mark to the right of the material selector

### Actual results

The Ultimaker homepage does not open.

### Expected results

Ultimaker-Cura-5.0.0-linux.AppImage
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaDXWL8l/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaDXWL8l/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
kde-open5: symbol lookup error: /lib/x86_64-linux-gnu/libKF5GuiAddons.so.5: undefined symbol: wl_proxy_marshal_flags

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Internet connection works, Firefox is installt

nallath, 2022-06-20:
Can you login? If not, it's likely the same issue as https://github.com/Ultimaker/Cura/issues/12291

x40-Community, 2022-06-27:
Sign in doesn't work
Marketplace works

Joeydelarago, 2022-06-29:
I'm going to close this issue as a duplicate of [#12291](https://github.com/Ultimaker/Cura/issues/12291).