Ultimaker/Cura #12551: Cura crashes on startup when Single Instance is true

jeferod83, 2022-06-18:
### Application Version

5.0.0-beta+1

### Platform

Windows 11

### Printer

Creality Ender 3 V2

### Reproduction steps

1) Open `Preferences`
2) Enable `Use single instance of Cura`
3) Close Cura
4) Open Cura

### Actual results

Shows a crash screen asking me to clear my configuration files

### Expected results

Cura starting normally

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

The first time that rises this error, a menu to clear my config files appeared, I got scared and ended up cleaning it by accident.
When it starts again, a fresh install was come, obviously, and I followed the step by step, connected Cura to my account, and synced all my plugins. I've restarted Cura, and 100% working. When I enabled the option to "use a single instance of Cura" on the preferences, after a restart, the crash occurs and the clear config files appeared again. The workaround for this problem was closing it, opening `cura.cfg` manually, setting to `False` the param `single_instance`, and starts Cura again.

The log file was not generated when the error raises and the crash screen opens. So, I believe that a empty log was not much helpful.

fieldOfView, 2022-06-18:
I believe this was fixed in the final 5.0 release. Could you please update and try again?

nallath, 2022-06-20:
It was.

closed 2022-06-20 by nallath