Ultimaker/Cura #12554: Impossible to leave the "rotate object mode" when entered via the secondary button on a Wacom tablet pen.

hugovangalen, 2022-06-18:
### Application Version

5.0.0

### Platform

Linux

### Printer

Ender3 Pro

### Reproduction steps

Use a Wacom tablet secondary button on the pen to rotate the object.

Stop pressing the secondary button to get out of rotate mode.

### Actual results

The object rotates, but it is impossible to escape rotate mode, bar restarting the application. Even using the 2nd mouse button which also enters/leaves rotate mode is useless at this point.

### Expected results

Leave rotate view mode when the secondary button on the pen is released. 

Note that this worked perfectly fine in 4.x versions.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No files to include. Nothing is logged when this occurs and this bug happens with an empty project.

3dHawwy, 2022-07-15:
Same problem here.  Mac, Wacom or touch pad.  Nothing works.

Ghostkeeper, 2022-07-15:
What mouse command is your second button mapped to? If it is not the 2nd mouse button, this is expected behaviour.

MariMakes, 2022-08-23:
Hey @hugovangalen,
Thanks for the report!

I literally dusted off my old Wacom Bamboo CTE 440 to verify your findings.
But it's not showing any unexpected behavior for me, perhaps because I'm on a windows device.

The issue however seems familiar to me. 
If you encounter it again, can you try to press Esc to leave Rotate Object Mode? 

In the meantime, I'll ask around to see if we have someone on the team with a tablet that can check what is going on.

hugovangalen, 2022-08-29:
> The issue however seems familiar to me. If you encounter it again, can you try to press Esc to leave Rotate Object Mode?
> 
> In the meantime, I'll ask around to see if we have someone on the team with a tablet that can check what is going on.

I have just tested to verify: hitting Escape does not resolve the issue.

(I also tested this with the latest 5.1.0 version, by the way. Same thing happens there.)

felderkwek, 2022-09-04:
I am having the same issue as well.

Sjullep, 2022-10-07:
I have this problem too (MacOS 12.6 and a Wacom Intuos Pro). 
I had the same problem with the program Gyroflow. The solution de Gyroflow devs came with was updating QT version to 6.4.0. After that Gyroflow works fine here, so perhaps this is also the solution for the Cura 5.x problem?

z0u, 2022-10-17:
Same here on Mac OS.

Cura 5.1.1
Wacom Pen & Touch CTH-490 version 1.0.0.0
Wacom drivers 6.3.46-1
Mac OS 12.6 / Intel

MariMakes, 2022-10-28:
@GraafSybren mentioned that he might be able to supply a screen share of what he was running into.
Do you mind sharing it here?

wchairy, 2022-11-02:
I'm having the same issue. Seriously, what the heck. I'm disabled and it's my only touch point into the system. I see that you're wanting a screen grab but of what something not working.

GraafSybren, 2022-11-02:
I did a screen recording. Issue appears on 00:05
This bug seems to appear random after a while..
 there is no specific trick I have found to reproduce it.

https://user-images.githubusercontent.com/73584612/199455474-9020dabe-39f8-462a-aa87-7fc1691e3832.mov

z0u, 2022-11-10:
Here's another screen recording, showing pen taps.

Black: Pen tip / left-click:
[image]

Green: Side button / right-click:
[image]

https://user-images.githubusercontent.com/1408875/200986819-57e78cb6-d9df-420e-92cb-bc1093f6420a.mov

Issue triggered at 0:24. Notice that, even though the button is released, it's still in rotate mode.

Tablet settings:

[image]

z0u, 2022-11-10:
Possible clue: it seems to happen most reliably after the pointer jumps to another part of the screen.

Both of the following recordings begin with a fresh instance of Cura.

Generally OK: not lifting the pen far from the tablet:

https://user-images.githubusercontent.com/1408875/200991668-0692c2a1-abca-4c1e-ac16-985fbd0c526a.mov

Generally fails: dragging after the pointer jumps:

https://user-images.githubusercontent.com/1408875/200991923-5ee8c86f-04f1-45c2-ad5f-05099cd29782.mov

Braintoe, 2023-01-22:
I followed the forward from issue #13707 which suits the behaviour on my computer (Linux Mint with a Wacom Intuos Pro PTH-651 as a mouse replacement) better.

Things I can add to this:

- Cura (freshly installed as I am in the process of switching to a new 3D printer) registers movements of the pen but no clicks (i.e. tapping with the stylus or using a stylus button) at all
- if I put the stylus away and use the multitouch function of the tablet with my finger, it behaves normally
- Cura is not the only sofware affected by this bug, the Teamviewer remote access client shows exactly the same behaviour. This tells me it has to be some library (or such?) Cura and Teamviewer share
- all other programs I use behave as they should and recognise the stylus clicks in the same way as they do with a mouse.

Braintoe, 2023-01-23:
Thanks to a little help from the xf86-input-wacom team, I could test the input with the `xinput test-xi2` terminal command. I compared a mouse click and a fingertip onto the tablet (both recognised by Cura) with a pen "click". This is what I got:

A click with multitouch (finger):

```
EVENT type 15 (RawButtonPress)
    device: 2 (10)
    detail: 1
    flags: 
    valuators:

EVENT type 16 (RawButtonRelease)
    device: 2 (10)
    detail: 1
    flags: 
    valuators:
```

A mouse click with deliberate motion while pressing the button to simulate pen behaviour:

```
EVENT type 15 (RawButtonPress)
    device: 2 (16)
    detail: 1
    flags: 
    valuators:

--------- numerous EVENT type 17 (RawMotion) events ----------

EVENT type 16 (RawButtonRelease)
    device: 2 (16)
    detail: 1
    flags: 
    valuators:
```

A pen vertically dropped some centimeters onto the tablet surface with the hand kept at a distance to minimise any disturbances yielded the following results:

```
EVENT type 15 (RawButtonPress)
    device: 2 (8)
    detail: 1
    flags: 
    valuators:
          0: 16324.00 (16324.00)
          1: 15667.00 (15667.00)
          2: 42964.00 (42964.00)
          3: 0.00 (0.00)
          4: 4.00 (4.00)
          5: -900.00 (-900.00)

--------- numerous EVENT type 17 (RawMotion) events ----------

EVENT type 16 (RawButtonRelease)
    device: 2 (8)
    detail: 1
    flags: 
    valuators:
          0: 15986.00 (15986.00)
          1: 15339.00 (15339.00)
          2: 0.00 (0.00)
          3: 0.00 (0.00)
          4: 2.00 (2.00)
          5: -900.00 (-900.00)
```

Could it be that whatever you use to get mouse events has problems handling these valuators that come with the button events?

Braintoe, 2023-01-24:
Small update to the commente above: I overlooked that I just copied the "Raw" events, sorry. In all cases above, there are also ButtonPress/Release events:

Pen:
```
EVENT type 15 (RawButtonPress)
    device: 2 (8)
    detail: 1
    flags: 
    valuators:
          0: 28385.00 (28385.00)
          1: 4290.00 (4290.00)
          2: 20457.00 (20457.00)
          3: 20.00 (20.00)
          4: 26.00 (26.00)
          5: -900.00 (-900.00)

EVENT type 4 (ButtonPress)
    device: 8 (8)
    detail: 1
    flags: 
    root: 2438.17/221.09
    event: 100.17/81.09
    buttons:
    modifiers: locked 0x10 latched 0 base 0 effective: 0x10
    group: locked 0 latched 0 base 0 effective: 0
    valuators:
        0: 28385.00
        1: 4290.00
        2: 20457.00
        3: 20.00
        4: 26.00
        5: -900.00
    windows: root 0x1e0 event 0x7400001 child 0x0

EVENT type 9 (FocusIn)
    device: 3 (3)
    windows: root 0x1e0 event 0x7400001 child 0x0
    mode: NotifyNormal (detail NotifyNonlinear)
    flags:  [same screen]
    buttons: 1
    modifiers: locked 0x10 latched 0 base 0 effective: 0x10
    group: locked 0 latched 0 base 0 effective: 0
    root x/y:  2438.00 / 221.00
    event x/y: 100.00 / 81.00

EVENT type 4 (ButtonPress)
    device: 2 (8)
    detail: 1
    flags: 
    root: 2438.17/221.09
    event: 100.17/81.09
    buttons:
    modifiers: locked 0x10 latched 0 base 0 effective: 0x10
    group: locked 0 latched 0 base 0 effective: 0
    valuators:
        0: 28385.00
        1: 4290.00
        2: 20457.00
        3: 20.00
        4: 26.00
        5: -900.00
    windows: root 0x1e0 event 0x7400001 child 0x0

EVENT type 17 (RawMotion)
    device: 2 (8)
    detail: 0
    flags: 
    valuators:
          0: 28382.00 (28382.00)
          1: 4289.00 (4289.00)
[... 193 lines cut ...]
    windows: root 0x1e0 event 0x7400001 child 0x7400002
EVENT type 5 (ButtonRelease)
    device: 2 (10)
    detail: 1
    flags: 
    root: 134.63/220.88
    event: 84.63/89.88
    buttons: 1
    modifiers: locked 0x10 latched 0 base 0 effective: 0x10
    group: locked 0 latched 0 base 0 effective: 0
    valuators:
    windows: root 0x1e0 event 0x7400001 child 0x7400002
EVENT type 8 (Leave)
    device: 2 (2)
    windows: root 0x1e0 event 0x7400001 child 0x0
    mode: NotifyUngrab (detail NotifyInferior)
    flags: [focus] [same screen]
    buttons:
    modifiers: locked 0x10 latched 0 base 0 effective: 0
    group: locked 0 latched 0 base 0 effective: 0
    root x/y:  134.00 / 220.00
    event x/y: 84.00 / 89.00
```

Again, the valutators are the only obvious difference for these events unless I overlook something. While most problems handle these events just normal, Cura and Teamviewer do not.

MariMakes, 2023-01-26:
Wow! That's some amazing feedback @Braintoe.
Super helpful ❤️ 

I'll bring it up with the team again. 🤞

Braintoe, 2023-01-26:
> Wow! That's some amazing feedback @Braintoe. Super helpful heart
> 
> I'll bring it up with the team again. crossed_fingers

Glad I could help! This problem is driving me crazy... 

One more thing I found: while I have no clue how to dismantle or replace content in the Cura appimage, I was able to extract the contents of the .deb installers of the Teamviewer versions. Thus, I installed the last version of that program which still accepted pen inputs and replaced files with then ones from the version directly after that one in order to narrow down a bit what might be the cause. (In opposite to Cura, Teamviewer is overall smaller and the problem was introduced during a minor upgrade which makes this approach easier.) 

The result I found - and that might help you even if it comes from an unrelated program - was that neither the QT libraries (QT 5, in their case) nor the XCB libraries Teamviewer comes with caused the error. The problem is somewhere within their main executable, so I hope this might be the case with Cura as well.

Braintoe, 2023-02-19:
Hi there, I just noticed one more thing regarding this issue:

if you click with the tablet pen onto an item such as an entry in the menu on the top left of the "settings" window or a material in the list in settings --> materials), this item gets stuck in a seemingly "semi-selected state" and becomes unusable even if you switch to using a mouse. You have to close and re-open the corresponding window to be able to access that item again. This might also happen to the rotate object mode the original bug report above is about.

This does not apply to all items though.
Here is a list of the items that become blocked and of the ones that stay accessible if I click onto the item with the pen and try to access it with the mouse afterwards (probably incomplete, but I hope it helps):

What becomes blocked against any later mouse interaction by a tap with the tablet pen:: 
- the "prepare / Preview / Monitor" buttons on the top
- the dropdowns for choosing the printer the material and the print settings
- the object rotation - it does not get released
- any items in the "plugins" or "materials" list in the Marketplace (switching to the Details view will become impossible)
- the "Close" button (i.e. the cross) in the blue hint boxes (tested with the hint "Manage your Ultimaker Cura Plugins..." hint in the Marketplace)
- the main menu on the top left of the "Settings" dialogue
- the different items in the lists showing printers, materials and profiles in the corresponding sections of the "Settings" dialogue
- buttons in confirmation dialogues (tested with the "No" and the "Yes" button in the "confirm remove" dialogue for materials)

What stays accessible:
- the File-Edit-View-etc dropdown menus
- the button on the top left which accesses the "file open" window
- the three buttons on the top right ("Marketplace", the button with the 3x3 dots and the "Sign in" button)
- if a dropdown for choosing printer, material or print settings happens to be already open, then any items such as dropdowns, text fields and checkboxes that are shown in there stay accessible as well
- the three buttons "Plugins", "Materials" and the toothgear button in the Marketplace
- the right part of the "Settings" dialogue (e.g. the "Information" and "Print settings" tab for a material including their content and the burger icon above these tabs
- the dropdown menu behind that burger button

If you want me to try any other items please do not hesitate to ask.

Sjullep, 2023-02-19:
Thanks for all the testing. These problems drive me nuts and are making Cura unusable for me...
Sometimes when items are "stuck", right-clicking with the pen a few times releases it. But sometimes it makes no difference....

Braintoe, 2023-03-11:
@MariMakes Do you happen to have any news about the issue?

Having to dig out a mouse just for Cura is really unnerving, especially since I tend to open a model in Cura and then tap into the 3D view - which means Cura gets caught in the 3D rotation with no means to escape from there, apart from closing Cura and start over.

In order to try to solve this issue, I would like to help as good as I can. If someone could simply explain to me where to find examples in the Cura code for the working and the dysfunctional mouse events noted above, I will be happy to compare these and try to find my way forward from there.

wchairy, 2023-03-13:
This is an accessibility nightmare. So many people use different types of
input methods. Myself being one of them and this makes cura unusable

MariMakes, 2023-03-20:
Hey All,

Quick update on what has been happening behind the scenes. 
At first, we thought changing this was outside of our control, since we are dependent on the QT library for fixing the bug on their side. But there seems to be a difference between TabletEvents, TouchscreenEvents, and MouseEvents that we are not yet utilizing.

After asking around we acquired a Wacom tablet in the office to test both the problem. I'm on Windows and was not able to reproduce it. But after handing it over to one of our Mac developers, were we able to reproduce it. So now we can see what's happening when the problem occurs and if we fix it if the fix works. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10420

Thanks for the report! 👍

Sjullep, 2023-03-20:
👍

Braintoe, 2023-03-20:
@MariMakes This is great news, thank you very much in the name of my right hand!!

As I noted before, the QT libraries do not seem to be the primary culprit here if I extrapolate what I found with Teamviewer (see [above](https://github.com/Ultimaker/Cura/issues/12554#issuecomment-1405410658)) - therefore I am very happy to see you found some further differences.

Did you also manage to recreate the issue on a Linux PC? If not, I hereby volunteer for testing if this helps.

wchairy, 2023-03-20:
Thanks so much

Sjullep, 2023-03-20:
Great news, hope it gets fixed soon now👍