Ultimaker/Cura #12555: Setting heating with M104 S[first_layer_temperature] ; start nozzle heating but don't wait not working CURA waits anyway

Sonicboom247, 2022-06-18:
### Application Version

5.0.0

### Platform

Windows 10 X64

### Printer

Artillery Hornet

### Reproduction steps

adding the below to start Gcode in Prusa works correctly and bed heats up and then printer starts tilting mesh, however in Cura printer waits for both temps to finish heating before moving.

; Set the heating
M190 S[first_layer_bed_temperature] ; wait for bed to heat up
M104 S[first_layer_temperature] ; start nozzle heating but don't wait

### Actual results

printer waits for both temps to finish heating before moving.

### Expected results

Bed should heat and printer should continue instead of waiting for hotend

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[NutCurved_v3.3.zip](https://github.com/Ultimaker/Cura/files/8933901/NutCurved_v3.3.zip)

GregValiant, 2022-06-18:
Those are "Keywords" for PrusaSlicer.  Cura's are different.  @fieldOfView posted the list of Cura keywords [HERE](http://files.fieldofview.com/cura/Replacement_Patterns.html).
I put this together using the StartUp Gcode of your Hornet.
;
; StartUp
;
G90  ;Absolute Movement
M82 ;Absolute Extrusion
G21 ;Metric units
M220 S100  ;reset feed rate
M221 S100  ;reset flow rate
M140 S{material_bed_temperature_layer_0}  ;start to heat the bed
M109 S150  ;heat the hot end to 150 and wait
M190 S{material_bed_temperature_layer_0}  ;wait for the bed just in case
G92 E0     ; Reset E
G28     ; Home XYZ axes
M117 Homing and Leveling
G29 A ;  Activate UBL
G29 L0 ; Load Mesh slot 0
G29 J3 ; Probe Bed for mesh tilting
G1 Z5 F600    ; move Z
G1 X0 Y0 F6000  ;move to corner to ooze as the hot end heats the rest of the way.
M109 S{material_print_temperature_layer_0}  ;heat the hot end to first layer temperature and wait for it
G1 X2 Y2 F2000; Move to start of priming line
G1 Z0.3 F600 ;drop to priming height
G1 X130 Y2 Z0.3 F1500 E20; Draw priming line
G92 E0 ;Reset extruder
M117 Printing...
;
;End of StartUp
;

This is a problem with your start up gcode rather than a Cura bug.  Because you have those incorrect keywords Cura is automatically adding M109 and M190 lines before your startup gcode.  You can try the above or just change your existing using Cura keywords and let us know if it works.  I'll remove the "bug" label for now.

Sonicboom247, 2022-06-18:
Fantastic, thank you, i will test this now

GregValiant, 2022-06-19:
You can't level the bed with the build plate and nozzle cold and then print with them hot because they both undergo some thermal expansion and change shape at least a little bit.  What I did there was to start the bed and the hot end at the same time.  The hot end heats to 150 but say you are printing PETG with the bed at 85.  The hot end would get to 150 a lot faster than the bed would get to 85 and the bed needs to be at the set temperature during leveling.  That's the reason for the M190 line.
You can do what ever you want with your StartUp Gcode.  I just posted one way (that will work) there are certainly others.

Sonicboom247, 2022-06-19:
Understandable but if the probe is the only thing touching the bed, not the nozzle then the only thing that needs to be at temp is the bed wouldn't it?  After G28, my nozzle is already at temp by the time UBL mesh tilting is complete. I also print PETG at 70, but that's irrelevant since it waits for bed to heat but not nozzle. No worries if Cura doesn't have a way to ignore  
{material_print_temp_wait} | Wait for Nozzle Heatup | Whether to wait until the nozzle temperature is reached at the start.

GregValiant, 2022-06-19:
Your Z offset can probably adjust for the expansion of the nozzle as it comes up to printing temperature from a cold state.  The nozzle expansion on my Ender 3 Pro has a lot more effect because the hot end mounts to the X carriage at the top of the aluminum heat exchanger.  Aluminum has comparatively high thermal expansion and in my case all that expansion is downward and I don't have an ABL.  I want mine to be hot when I level so I get it right.

Sonicboom247, 2022-06-19:
Correct,  this was the fact when I did not have a probe and was manually bed leveling,  I fitted my Hornet and one of my Sidewinders with probes so once a print starts I usually only have to adjust probe offset once and then I am good for a few months. then I like to do a little maintenance on the printers and make a new mesh and start fresh.  as much as I print I figure I am wasting about an hour or two a month just waiting on the hotend to heat up..

so at this point is there a proper translation that Cura will process that will not wait for hotend temp or is this an actual bug?

GregValiant, 2022-06-20:
Replace the M109 with M104.  This is not a Cura thing but a firmware thing.  Cura insures that there won't be a cold extrusion when starting a print by inserting the heating commands if there are no keywords in the StartUp gcode.  If there are keywords regarding the temperatures then the StartUp Gcode is an over-ride available to any user. There is no bug here.  You want a custom StartUp then you need to write the gcode lines in a way that your firmware understands and that does what you want.  Once you put temperature keywords into your Start gcode it is up to you to get it to get the code right because Cura is out of the loop except for passing your StartUp gcode to the actual gcode file.
A reason Cura uses M109 and M190 is because some power supplies can't handle the load of both the hot end and bed powering up together from a cold state.  All that current goes through the mainboard.  Once they are up to temp and floating it isn't a problem.

Sonicboom247, 2022-06-20:
i think that is what the title is about, also just double checked with the below and it still waited, same exact command in Prusa does what i expect
; Set the heating
M190 S[first_layer_bed_temperature] ; wait for bed to heat up
M104 S[first_layer_temperature] ; start nozzle heating but don't wait

Sonicboom247, 2022-06-20:
Thank you, this is working perfectly in PrusaSlicer without issue so I will continue to use Prusa
M190 S[first_layer_bed_temperature] ; wait for bed to heat up
M104 S[first_layer_temperature] ; start nozzle heating but don't wait