Ultimaker/Cura #12558: Material Color Picker

Slack3D, 2022-06-19:
### Is your feature request related to a problem?

In the new Cura update 5.0 the material color picker is suddenly very limited. There was a color wheel that gave you a range of each color throughout the spectrum in the 4.x.x version I was on before the update. Is there a way to add the color wheel back to the color picker?

### Describe the solution you'd like

I would like a color wheel in the materials menu when making/editing existing color information and settings. 

### Describe alternatives you've considered

If a color wheel is not an option, I think there should be more options for colors. Currently there are 20 color options. With new filament technology, filament can be a wide range of colors, the program used to slice a model with exact material information should reflect this. 

### Affected users and/or printers

Any user/printer that utilized the current version of Cura

nallath, 2022-06-20:
Due to the upgrade to Qt 6, we lost the colorpicker element that we used (as this hasn't been ported over yet). As such we quick whipped something together to at least have something.

You can at least set it to be any color that you want by copying the hex color into the textfield. 

I will discuss this with the rest of the team!

SGT-276, 2023-10-08:
Is it possible to add an icon for rainbow-colored filament?