Ultimaker/Cura #12560: Directx Crashing

Lrt19, 2022-06-19:
### Application Version

5.0, 4.x

### Platform

Windows 10

### Printer

Voxelab Aquila X2

### Reproduction steps

Slicing anything, then switching from a 3d preview then back to prepare, then slice again, preview, crash.

### Actual results

Usually just crashes the program, restart, but last time it blue screened.

### Expected results

not crash.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

log and dxdiag and bluescreenview output included
[bluescreenview.txt](https://github.com/Ultimaker/Cura/files/8934858/bluescreenview.txt)
[cura.log](https://github.com/Ultimaker/Cura/files/8934859/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/8934861/DxDiag.txt)
.

Asterchades, 2022-06-20:
I'd be inclined to suspect graphics driver corruption. Cura doesn't even use DirectX for rendering - it uses OpenGL, which your graphics drivers will be responsible for either handling directly or wrapping to DirectX. My first instinct would be to remove the existing display drivers using [DDU](https://www.wagnardsoft.com/display-driver-uninstaller-ddu-), then download and install the [latest available drivers](https://www.amd.com/en/support/graphics/amd-radeon-hd/ati-radeon-hd-5000-series/ati-radeon-hd-5870) (it says HD 5870 but that should cover the whole HD 5800 series) unless you have a specific reason to use an older version.

It may also be that the OpenGL portion of your display drivers simply isn't entirely compatible with what Cura uses. In that case you can use the "Force layer view compatibility mode" option in the Cura preferences to use OpenGL 2.0 instead of OpenGL 4.1. It isn't as full featured but the older standard should be more stable.

Give one or both of the above a try and get back to me.

Being that we're talking about a potentially 12-13 year old graphics card, however, there's also the possibility of failing hardware.

Lrt19, 2022-06-24:
Thank you for the reply. I do have the latest drivers for this card, it is rather old and is only found under the legacy graphics downloads. I'm turning on the layer view compatibility, to give that a try, though considering the inferior image quality, I hope that's not the only answer.

Joeydelarago, 2022-06-29:
Hello, can you tell us if using "Force layer view compatibility mod" worked for you?

Lrt19, 2022-07-17:
Yes the compatibility mode seems to be perfectly stable, though the quality of the image is so terrible, I can't use it this way.  I'll be going back to the regular default mode, and will just have to live with the instability.

closed 2022-07-17 by Lrt19