Ultimaker/Cura #12564: Viewport Navigation Issues

Fordari, 2022-06-20:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

NA

### Reproduction steps

New Surface Pro running Windows 11 has issues with navigation. You would expect to be able to rotate when holding right click but sometimes it doesnt work. Only after hitting escape on occasion does it function. Sometimes the keys work to rotate around the model, other times they do not. 

### Actual results

Hit or miss functionality using trackpad. 

### Expected results

Keys should always work as navigations, up down forward back, and if you hold shift, you should be able to translate left right up down. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NA

nallath, 2022-06-20:
Unfortunately this is not something that we've been able to reproduce. As there also haven't been any other reports of this, there is little that we can do to figure out what is causing it.

In the past I've have seen cases where issues with the mouse controls were caused by software that adjusted the theming of your system. Perhaps something similar is happening with you?

Fordari, 2022-06-21:
It is not. When I have a mouse connected, Cura viewport navigation works fine. Immediately after using the mouse, the trackpad functions as you would expect, then pan functionality ceases to work. 

It may be a Windows 11 issue, so I'll have to install a compatible version (if any exists). The root of the issue may lie in the lack of touch functionality. Cura is unusable without a mouse in the current state, at least on Surface devices out-of-the-box. 

I would be happy to upload a video if you'd find it worthwhile.

Fordari, 2022-07-03:
Please see the uploaded video. You can see that panning works on the first click, then the only viewport navigation that functions after is zoom. No touch functionality works besides editing the object. 

https://user-images.githubusercontent.com/50779811/177048786-4818bc2a-1028-42c9-b19b-ff00b3508be9.mp4

nallath, 2022-07-05:
I have a lenovo thinkpad whch still has physical left / right mouse buttons. That's probably why I couldn't reproduce it. For me I can always use the right mouse button even without a mouse.

Fordari, 2022-08-16:
I was hoping 5.1 solved the issue, but Cura is still unusable without a mouse on Surface for me.

MariMakes, 2022-08-23:
Hey @Fordari,
Thanks for the report.

Are you encountering similar issues mentioned in  https://github.com/Ultimaker/Cura/issues/2515?
We haven't fully unlocked touch screen support, yet. 
If so can you let us know, so I can close this issue as a duplicate?

In the meantime I'll bring it up with our team, they might know where to start troubleshooting this.

closed 2022-09-10 by github-actions[bot]