Ultimaker/Cura #12565: Search Bar Font in Print Settings

mike-pisman, 2022-06-20:
### Application Version

 5.0.0

### Platform

Linux (Ubuntu/Debian based)

### Printer

Custom

### Reproduction steps

Hi, I have a 4k monitor and was working on custom theme for Cura to scale fonts and icons plus change the colors to my liking. Based on config provided bellow, almost everything works except the font in the search bar of the Print Settings. Does anyone know which font class is used for the search bar? 

[theme.txt](https://github.com/Ultimaker/Cura/files/8937637/theme.txt)

```{
    "fonts": {
        "default": {
            "size": 2,
            "weight": 400
        },
        "default_bold": {
            "size": 2,
            "weight": 600
        },
        "default_italic": {
            "size": 2,
            "weight": 400,
            "italic": true
        },
        "small": {
            "size": 1.75,
            "weight": 400
        },
        "small_bold": {
            "size": 1.75,
            "weight": 600
        },
        "small_italic": {
            "size": 1.75,
            "weight": 400,
            "italic": true
        },
        "small_emphasis": {
            "size": 1.75,
            "weight": 700
        },
        "medium": {
            "size": 2.25,
            "weight": 400
        },
        "medium_bold": {
            "size": 2.25,
            "weight": 600
        },
        "medium_italic": {
            "size": 2.25,
            "weight": 400,
            "italic": true
        },
        "large": {
            "size": 2.5,
            "weight": 400
        },
        "large_bold": {
            "size": 2.5,
            "weight": 600
        },
        "large_italic": {
            "size": 2.5,
            "weight": 600,
            "italic": true
        },
        "huge": {
            "size": 2.75,
            "weight": 400
        },
        "huge_bold": {
            "size": 2.75,
            "weight": 600
        },
        "huge_italic": {
            "size": 2.75,
            "weight": 600,
            "italic": true
        }
    }
}```

### Actual results

Search bar font is still small
![cura_screenshot](https://user-images.githubusercontent.com/58354623/174528320-7d49d945-f3ff-4f82-98b4-f8e52493e311.png)

### Expected results

Search bar font use small-italic styling

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Github does not allow .json, so I uploaded the theme in .txt

nallath, 2022-06-20:
The filter is created here: https://github.com/Ultimaker/Cura/blob/f474381425112f32ff190629bf0035e23fabd1e8/resources/qml/Settings/SettingView.qml#L40

It uses textfield (https://github.com/Ultimaker/Cura/blob/main/resources/qml/Widgets/TextField.qml) for it, which sets the font to be "default"

mike-pisman, 2022-06-20:
Thanks @nallath, I took a look. You are right, the font should be changed according to the default settings; Perhaps, it's the height or the top padding that affects the text. Do you know if I can change those setting in the config or would I have to build cura from source in order to changer those setting?

commit 52ea80f6 (2022-06-21): Fix the font for searchbars being styled correctly

By setting the font.italic to true, we actually reset the entire font.

Fixes #12565
closed 2022-06-21 by nallath