Ultimaker/Cura #12566: ultimaker Cura 5 menus are very slow and not working normally.

macrosat, 2022-06-20:
### Application Version

5,0

### Platform

Windows 10

### Printer

Trivend

### Reproduction steps

I want to open menus.

### Actual results

very slowly and not work normally.

### Expected results

ultimaker Cura 5 menus are very slow and not working normally.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

ultimaker Cura 5 menus are very slow and not working normally. #

nallath, 2022-06-20:
Could you please download the extensive logging plugin and run it. It will create an extra file that might show us why things are so slow with you. You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging

macrosat, 2022-06-22:
[remote-support_22-06-22-22_41_15.zip](https://github.com/Ultimaker/Cura/files/8960854/remote-support_22-06-22-22_41_15.zip)
extension menu not working with mouse

3Duke3, 2022-07-15:
There was a windows update last night on my PC, navigating Cura became very slow and unresponsive afterwards, especially switching between move/scale/rotate ect. Restarted PC, restarted Cura, no change. Found this ticket and installed the Remote Support plugin, after enabling CPU logging and restarting Cura the problem has gone away.