Ultimaker/Cura #12569: STL File Loading Problem

plainpylut, 2022-06-20:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 5 Pro

### Reproduction steps

1.  Start Cura
2. Load "ApprenticeS15e_Cowl.stl
3. STL crashes CURA or STL is Yellow and Grey Stripped.  It appears much larger than it should.
4. Hit Control Z if Cura doesn't crash and it is normal size.
5. Debug zip and STL file attached.

### Actual results

Cura sometimes crashes or the STL is displayed extremely large.  Hitting Control Z usually will restore the STL to correct size.
[remote-support_22-06-20-15_28_00.zip](https://github.com/Ultimaker/Cura/files/8944015/remote-support_22-06-20-15_28_00.zip)
[ApprenticeS15e_cowl.zip](https://github.com/Ultimaker/Cura/files/8944208/ApprenticeS15e_cowl.zip)
[CE5P_Apprentice_Cowl.zip](https://github.com/Ultimaker/Cura/files/8944504/CE5P_Apprentice_Cowl.zip)
[Roaming_cura_5.0.zip](https://github.com/Ultimaker/Cura/files/8944527/Roaming_cura_5.0.zip)

### Expected results

STL should load normally.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

See attached files in field "Actual Results"
[Roaming_cura_5.0.zip](https://github.com/Ultimaker/Cura/files/8944537/Roaming_cura_5.0.zip)
[CE5P_Apprentice_Cowl.zip](https://github.com/Ultimaker/Cura/files/8944547/CE5P_Apprentice_Cowl.zip)

GregValiant, 2022-06-21:
_"STL is Yellow and Grey Stripped. It appears much larger than it should."_
I opened the STL and checked the model for errors (which Cura shows in the model as different colored areas) and there were no errors and the slice looked good.
I opened the 3mf and same thing.  The model looked good and the slice looked good.  (I don't think it would print successfully without infill though.)
In the 3MF the model is 129.04 x 111.36 x100 tall but the STL loaded with the model at 128.04 x 110.36 x100 tall.  So the X and Y are off by 1mm and the Z dimensions match.  That is odd.
Someone else will need to take a look at the log files.

nallath, 2022-06-21:
In the 3MF the model is simply saved to be bigger. 

[image]

I don't see any real difference between the two models here. I'm also unable to make it crash.

plainpylut, 2022-06-21:
Nallath,
Yes, I scaled the cowl 1mm in the X and Y direction(not uniform).  It needs to be printed with 0% infill, No top layer and 2 bottom layers.
When I load the project file (3mf) it appears normal.
Here is how to duplicate my issue:
1.  Load the project file (CE5P_Apprentice_Cowl.3mf).
2. Delete the STL file (object).
3. Open the Apprentice_Cowl.stl file.  As soon as you see the Cowl displayed, scroll the mouse wheel.
The STL file increases in size.  The only way to recover is hit "CTL Z".  Sometimes Cura totally exits.
4. I'm running a HP ZBook 15 (I7 processor, 2.70 GHz) with 16G RAM with latest drivers from HP.
5. I tried removing the Barbarian plug-in; no change after restarting Cura.
6. Attached are proj
[CE5P_Apprentice_Cowl.zip](https://github.com/Ultimaker/Cura/files/8952811/CE5P_Apprentice_Cowl.zip)
ect, STL and two screen captures.

GregValiant, 2022-06-22:
I followed the reproduction steps.  Everything appears to work as I would expect it to.  Immediately scrolling the mouse wheel when the second STL shows up results in zooming in and out with no change in the size of the model.
I repeated the steps with the camera view in both perspective and in orthographic and there was no change in the model.
This laptop is in compatibility mode and I'm unable to reproduce the problem..

plainpylut, 2022-06-23:
Ok.  I uninstalled, rebooted, and re-installed Cura 5.0.0.  Everything seems fine now.  Something must have gotten corrupted!
Please close #12569.
Thank You!

plainpylut, 2022-06-23:
I uninstalled, rebooted, and re-installed cura 5.0.0.  It is working fine.
Please close #12569.
Thank you!

closed 2022-06-23 by plainpylut