Ultimaker/Cura #12572: brim z-offset

dschil138, 2022-06-21:
### Is your feature request related to a problem?

Sometimes prints don't adhere to the bed well. Brims can help but don't always work. If you use z-offset to squish your first layer down, you get worse elephant's foot.

### Describe the solution you'd like

Sometimes I will do this manually if I want better brim adhesion but want to avoid elephants foot. I will set the brim to ten lines thick, and for the first few I will manually adjust the z-offset on the printer to be -0.04. Then, as it circles in closer to the print, I will increment it back to -0.03, -0.02.... until the first wall of the actual print starts and we are back to 0.0.

It would be really nice to have a setting to do this automatically! Even if it was just a simple brim-specific offset to do the whole thing at like -0.03, I think it would be helpful!

### Describe alternatives you've considered

Alternative is doing it manually, which is fine but kind of a pain if you are doing a bunch of prints.

### Affected users and/or printers

users with adhesion & elephant's foot issues

fieldOfView, 2022-06-23:
Hi, I think this could be useful so it's worth the experiment. I will add this to the next version of the Z Offset plugin.
https://github.com/fieldOfView/Cura-ZOffsetPlugin

Joeydelarago, 2022-06-28:
If you are having adhesion issues, then you should try using a textured build plate or maybe getting your bed more leveled. 

This would be a cool feature, but I don't think we have the time to implement, unfortunately. I will defer it for now and hopefully someone in the community will work on this functionality.

5axes, 2022-07-20:
> I have added this feature to my Cura variant and it will be in the next release (not yet scheduled)

@smartavionics the 4.20.1 is your "next" release or you are talking about an other one ?

smartavionics, 2022-07-20:
Hi @5axes , It's going to be in 4.20.2 which I will release in the next few days.

5axes, 2022-07-22:
Hello @smartavionics ,

4.20.2 Brim Z offset works fine, I've also made some test with the Plugin of @fieldOfView to check if they are no issue it's ok Even in Extensive processing mode the global Z Offset added by the plugin is changing your Z Offset values.  I don't know if a "Progressive Z Offset" as describe by @dschil138 could be integrated but it lokks like a great idea. 

Thanks for your job @smartavionics

haytham2004, 2022-11-14:
I am having the same issue when using the Brim only. I haven't used brim in a while and just started using again in the past few years and my z-offset is all over the place. I have to lower the nozzle for the brim then I have to raise about 0.5 - 1 mm for it to print fine. I tried using the z-offset setting in Cura but the problem seems to only happen when printing the brim. I of course did the usual diagnosis of checking everything but and narrowed it down to the brim. The one thing I am hesitant about using is increasing the flow when the brim print. I am worried this will clog the nozzle.

closed 2023-11-29 by github-actions[bot]

5axes, 2023-11-29:
Still not present in Cura 5.6