Ultimaker/Cura #12577: 'zoom toward mouse direction' not working properly on high resolution machine (3840 x 2160)

discip, 2022-06-21:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

not relevant

### Reproduction steps

I am experiencing a strange behavior of the 'zoom towards mouse' feature.
It does not work as it did before Cura 5.0.0.
Although zooming works, the movement does not keep up.

This issue is not related to a specific printer.

Since I am using Cura on 2 different machines and the bug occurs only on the machine with the higher resolution (3840 x 2160), I assume the higher resolution to be the cause.

### Actual results

No proper zooming to wards mouse.

### Expected results

Movement should keep up with the zoom.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8950299/cura.log)

nallath, 2022-06-21:
Fixed by: https://github.com/Ultimaker/Uranium/pull/817

discip, 2022-07-03:
@nallath
Thanks! 👍

discip, 2022-07-11:
@nallath
In the beta 5.1.0 this is still not working!

nallath, 2022-07-13:
Well, yes, because it wasn't merged yet ;)

discip, 2022-07-13:
Ah, I see.
Thanks.

discip, 2022-07-22:
@nallath
Still not fixed for latest stable release (Cura 5.1.0).

MariMakes, 2022-08-23:
Hey @discip
It seems like the fix was not merged in time for release. 😢 
I'll go ask the team what happened and if we can have it for the next release instead. 
Fingers crossed 🤞

discip, 2022-08-23:
@MariMakes
### Thank you. 😃

discip, 2022-10-04:
Seems to be fixed in 5.2.0-beta!

closed 2022-10-04 by discip

MariMakes, 2022-10-04:
[image]