Ultimaker/Cura #12579: Slicing issue

GabrielStevenson, 2022-06-21:
### Application Version

5.0.0

### Platform

10

### Printer

Enderv2

### Reproduction steps

1. I slice a model
2. Extract the model
3. print the model

### Actual results

1. couldnt slice the model

### Expected results

that I could slice the model

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

GabrielStevenson, 2022-06-21:
number 10 means windows 10

GabrielStevenson, 2022-06-21:
[image]

GregValiant, 2022-06-22:
1. You use Windows 10.
2. You are using Cura 5.0.0.
3. A model didn't slice.
With the model loaded in Cura use the "File | Save Project" command.  That will create a 3mf project file.  Zip the file and post it here.  There is a reason the form you filled out asks for a Project File and the Cura.log file.

GabrielStevenson, 2022-06-22:
[New folder (2).zip](https://github.com/Ultimaker/Cura/files/8958151/New.folder.2.zip)

GregValiant, 2022-06-22:
And after all of that it sliced fine for me.
[image]

The next step is:  In Cura - under the Help menu the top command is "Show Configuration Folder" click on it.  In that folder is the Cura.log file.  Zip that file up and post it here.  The model doesn't appear to have any errors.  Hopefully whatever happened got logged in that file.

GabrielStevenson, 2022-06-23:
[Log.zip](https://github.com/Ultimaker/Cura/files/8966920/Log.zip)

cht685, 2022-06-23:
Did you ever get this to work? This just started happening to me this morning. I can load the files up in 4.13.1 and slice it just fine. But if I try to slice them in 5.0.0, I get this same error...I just opened another ticket for the same issue.

GabrielStevenson, 2022-06-24:
Hello I actually did the creator of the model your looking at sent it to me again and I dont know what kind of magic he did but its works now but I left this ticket open to see what happens

Joeydelarago, 2022-06-28:
> Hello I actually did the creator of the model your looking at sent it to me again and I dont know what kind of magic he did but its works now but I left this ticket open to see what happens

Hello, I've had a look at the logs. It seems that the original model you were using had both overlapping and disconnected faces which causes the slicing to fail. The creator probably loaded the mesh up in blender and cleaned these up. 

I'll close ticket since it is resolved.

closed 2022-06-28 by Joeydelarago