Ultimaker/Cura #12582: Uninstall key is written to user profile (should be written to hklm)

Angelworks, 2022-06-21:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker

### Reproduction steps

This is an issue most probably won't run into - but if you install this app as a different user its not easily uninstallable by another user.

In my case I'm installing it as system via configmgr (memcm) and the uninstall seems to get written to 

HKEY_USERS\.DEFAULT\Software\Microsoft\Windows\CurrentVersion\Uninstall\Ultimaker Cura 5.0.0

It really needs to be written to:

HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall

Fwiw - previous versions of Cura handled this correctly.

### Actual results

Uninstall key gets written to whatever user profile your logged into - if its system it gets written to the default user profile and copied to the user profile during login.

### Expected results

Since the app is installed to C:\Program Files - its kinda breaking the rule that the uninstall information should be written to the system and not the user profile.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't really have a log to show for this :(.

Ghostkeeper, 2022-06-28:
Yeah, that's an issue. Maybe not an issue for many people though. We've always put the registry keys there, and it's the first time this has come up as far as I can see.

It's probably caused by this line here: https://github.com/Ultimaker/cura-build-environment/blob/33db8b865b5694f4dc47c83790f7fac6c8bd6007/packaging/NSIS/Ultimaker-Cura.nsi.jinja#L14

I'll discuss it with my colleagues and see what the priority of this would be.

Angelworks, 2022-06-28:
Sounds like it - keep in mind this is mostly an issue for enterprises who roll this software out programatically (which is the issue in my case).

rburema, 2022-07-08:
> We've always put the registry keys there, ...

That turns out to be technically true, but not the complete picture even. We used to put most keys in `HKCR` (which is a shorthand for _both_ the local machine _and_ the current user). There was only _one_ key that used to be `HKCU` (which is _only_ the current user), which had to do with (as far as I can sus out) file-associations.

I think its safe to say that we can change that one-line definition 'back' to `HKCR`, I'll try it out in a branch. Not sure if I dare to touch that currently if it turns out _not_ to work though ... we're in the middle of overhauling the build-system after all.

rburema, 2022-07-19:
Seems this will not make it into 5.1 :-/

opsi-archIT-KIT, 2022-07-21:
No, same behavior in 5.1.0

closed 2022-07-21 by jellespijker

rburema, 2022-07-22:
Fix is now merged, so will be in 5.2 at least! :-)

opsi-archIT-KIT, 2022-09-21:
With 5.1.1 the registry entry in HKCU is gone.

But no registry entry in HKLM was created.

rburema, 2022-09-22:
Yes, that's because we base 5.1.x on the 5.1 branch, a lot of fixes will not be in a minor release.