Ultimaker/Cura #12586: Fractured surface which should be flat and smooth

Holly-Marie, 2022-06-22:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality CR-10 (Duet 3D V2 mainboard)

### Reproduction steps

I'm slicing my model with 0.4 mm which mostly works fine.

The model:
[TabelTennisHolder.zip](https://github.com/Ultimaker/Cura/files/8958293/TabelTennisHolder.zip)

### Actual results

The surface of layer 241 is fractured, has cut-ins and one line is missing altogether

[image]

### Expected results

Layer 241 has one smooth flat surface on the arc

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[ProjectAndModel.zip](https://github.com/Ultimaker/Cura/files/8958299/ProjectAndModel.zip)

GregValiant, 2022-06-22:
This is the second one of these in the last couple of weeks.  In that first case I was able to reproduce this with my own model (which was hockey puck so it was flat).  @Holly-Marie with your model - changing the Initial Layer Height to .21 fixes it.
@Ghostkeeper mentioned another setting that might be involved.  I can't remember the disposition of that other bug report and I'm unable to find it.

Joeydelarago, 2022-06-28:
Hello, this duplicates [this issue](https://github.com/Ultimaker/Cura/issues/12425) so I will close it. If you are looking for a workaround, there are some solutions mentioned in that issue.

closed 2022-06-28 by Joeydelarago