Ultimaker/Cura #12587: Dual Mixing Extruder issues

BigJBehr, 2022-06-22:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Geeetech A10M

### Reproduction steps

Loaded two stl files that when merged yield a two color Benchy.
Went through the process of selecting and merging the stl files.
Sliced it.
Tell my Geeetech A10M to print the two color Benchy.
Everything goes well until I get an error tone and the message "Printer Failed E1".
The A10M has two extruders that feed a single nozzle, mixing hotend.
Why is it trying to heat a hotend that does not exist (E1, only E0 is valid) ?
You can download the files here; https://www.thingiverse.com/thing:763622

### Actual results

The print ends with the error message "Printer Failed E1"

### Expected results

Print with no failures due to E1 not heating.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[TwoColorBenchyonA10M.zip](https://github.com/Ultimaker/Cura/files/8960936/TwoColorBenchyonA10M.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8960943/cura.log)

GregValiant, 2022-06-22:
You need to load the Printer Settings plugin from the MarketPlace.  There are two settings in it that you need. "Extruders Share Heater" and "Extruders share Nozzle".  If those are not enabled then Cura assumes 1 heater for each extruder.
"Extruders share Nozzle" brings up another setting "Shared Nozzle Initial Retraction" where you will need to enter a value so prints start correctly.

BigJBehr, 2022-06-23:
Thanks for the quick response. I added the plug-in and checked the two items you suggested. I re-sliced the same Benchy project with the new settings. A purge tower was created, first time I have seen that. However, at about a height of 7.5mm, the print failed due to the same E1 heating failure. Printer is an A10M, two into one, mixing hotend. There is no E1, hence the failure.

GregValiant, 2022-06-24:
With the model loaded into Cura use the "File | Save Project" command.  That will create a 3mf file.  Zip that file and post it here and I'll take a look.  It's a setting though as I've gotten the Geeetech multi-extruders to work with Cura.  It's just a question of "Which setting".

closed 2022-07-08 by github-actions[bot]