Ultimaker/Cura #12588: Update/sync

maccnh, 2022-06-22:
### Application Version

5.0.0

### Platform

Windows 10 PRO

### Printer

Cr-10S Pro

### Reproduction steps

Updated to 5.0.0

### Actual results

Try to sync/Update

### Expected results

Lockup. 
Not responding 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8961368/cura.log)

MariMakes, 2022-08-23:
Hey @maccnh,
Thank you for opening your first issue with us.  🎉 
Sorry, it took us a while to get back to you. 😢 

If I'm browsing through your logs I see it might be related to USB printing. 
But unfortunately, I do not have enough information to troubleshoot with. 

Are you still stuck? 
- If you are it helps if you formulate your problem as: 
  **I see ... but I expected ...**
  At least some steps we can perform to reproduce it on our sie. 
 Maybe you can include a video of what you are encountering instead?
- If it's resolved, can you let us know so we can close the issue? 🤞

maccnh, 2022-09-05:
Still will not update.

MariMakes, 2022-09-13:
Hey @maccnh,

Unfortunately, this is not information for us to troubleshoot. 😞 
Can you share what you are trying to update?

closed 2022-09-27 by github-actions[bot]