Ultimaker/Cura #12589: Merge STL's

paulsartory, 2022-06-23:
### Application Version

5

### Platform

window 10

### Printer

none

### Reproduction steps

Multiple STL's not merging and aligning correctly

### Actual results

Multiple STL's not merging and aligning correctly

### Expected results

Multiple STL's not merging and aligning correctly

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Multiple STL's not merging and aligning correctly

fieldOfView, 2022-06-23:
Perhaps if you said the same line a couple of times more, we will be able to reproduce your issue and fix it in a future version.
Or, you could actually help us help you, and include what you are trying to do, what application you used to create the models, and post a link to some sample models that fail to merge.

paulsartory, 2022-06-23:
I raised an issue in good faith and you feel the need to criticize me!!! Clearly people skills are not your strong suit.

This is the first time I have raised a bug report. I seemed to require a lot of information that did not appear relevant to the issue that was "failure to align merged STL's"
I was unable to merge and align all of the parts in this model.

Multi-Color Kidney Model by MosaicManufacturing

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Multi-Color Kidney Model by MosaicManufacturing

This model is designed to use color and illustrate an internal view of a kidney. We hope this project inspires y...
 |

 |

 |

Please don't respond in the same insulting manner.
Paul Sartory

GregValiant, 2022-06-23:
@paulsartory According to your original post there is no bug here.
You stated:
**Expected results**
Multiple STL's not merging and aligning correctly

And it appears from your line in the **Actual Results** that what is _**Expected**_ is exactly what was achieved.

The form here is designed to provide information to people like @fieldOfView (a main contributor to Cura and part of the "community" that maintains the project) so bugs can be chased down and fixed.  Without a clear and concise description of the steps you took, the problem that occurred, and what was expected (as opposed to what actually happened) it is impossible to put together an investigation.

The inclusion of a project file is necessary so the model, the printer, and all the Cura settings are available as tools to chase down a problem.
The log file is different tool and is also necessary.
Screen shots of what happened provide a graphic description of a problem.  A picture is still worth a thousand words.

I am inclined to close this because as of right this minute (and according to the information you provided) I don't see a bug here.  I'll leave it open in case you would like to provide ANY information regarding a problem you might be experiencing.

Not everyone is as gregarious as I am (it's part of my name) but a total of 7 words simply repeated 4 times??  _C'mon man!_

paulsartory, 2022-07-07:
I presume that you consider this a legitimate issue and would like to resolve it?
If that is the case, please tell me what information you want that is relevant to the issue and I will endeavor to supply it.
Please bear in mind that I am not an expert in project files, data logs etc just a hobbyist that uses the software to slice STLs.
RegardsPaul Sartory

GregValiant, 2022-07-08:
With your models loaded into Cura but not yet merged, and with the Cura settings the way you would want to slice it, select the "File | Save Project" command.  It is a sort of "save-as" that will put together a 3mf file with your printer, the model(s), and all your settings present inside it.  One more step...GitHub requires Zip files so in File Explorer right click on the 3mf file you created.  There should be a menu item "Send To..." and select "Compressed (Zip) Folder".  Post that zip file here.

I am also just a hobbyist.  I am well experienced as a CAD designer and have a good working knowledge of gcode.  Beyond that I am retired and chasing down these sorts of problems has added immensely to my knowledge of 3d printing.
All the bug reports here are treated seriously but nothing can really be done without knowing what the poster was doing, details about the model (I always check them for errors), and how Cura was set up, and what printer is involved.  My experience here is that many of the issues reported are actually model problems.  Second place goes to user error, third place would be actual Cura bugs, and in fourth place would be problems in printer definition files, and in fifth place would be really dumb combinations of settings the some people insist on using because they saw something on YouTube.
Cura 5.0 in particular has a string of bug reports where there is a problem slicing.  It seems to have been fixed in 5.1beta but we will see.  It is a complicated program and a simple change over here in the code can cause hidden breakage over there in other code.
This problem where merging doesn't seem to be working right is different.  Finding out what is really going on requires an investigation where all the relevant material is gathered and gone through.  You are holding all the relevant material.  The investigation is necessary to fix the problem no matter where it is.

paulsartory, 2022-07-13:
I have attached a simple 2 STL cube that merges correctly in the Arachne Beta version but not in Cura 5

[CCR10S_TestCube2ColorsA.zip](https://github.com/Ultimaker/Cura/files/9106485/CCR10S_TestCube2ColorsA.zip)

GregValiant, 2022-07-13:
Thank you.
@fieldOfView would you care to do the honors and see if there is an issue with the merge function?  The OP has the model files as STL's.

fieldOfView, 2022-07-14:
I think that this issue is up to how the two original STLs are created/saved. In otder for the "merge models" to work, the two STL models need to be saved in such a way that they share a common origin. The "merge models" function puts the models back together in the relative position they had in the program you used to created the STLs. If you move the two models apart in the modeling application before saving as STL, this won't work.

What application did you use to create the original STLs? 
In this case it will also be useful to see the original STLs, instead of the project file, so we can confirm that the problem does indeed lie in the STLs, and so we can test if there is indeed any different between the Arachne beta and the 5.0 release.

Please note that though I could have been friendlier in my initial response, both of the things I ask before I also asked in my original response.

GregValiant, 2022-07-15:
I opened your 3mf file and attempted to merge the models and the merge failed.
I then downloaded the models from Thingiverse and imported A and B into Cura.  The merge worked.
The merge also worked for the other 2 pairs of models.
In the image below I opened your 3mf file.  Then I brought in the A and B models that I had downloaded from Thingiverse.
I selected your models and attempted to merge them.  The merge failed again.
Then I selected the "fresh" models I brought in and you can see that the merge was successful.  They ended up at Z=-2.5 and that needed to be addressed but the models properly merged.
My conclusion here is that something happened to the models you imported and they no longer have "location" data attached to them.
As @fieldOfView says, the STL (or 3MF model file, or OBJ object file, etc.) must have location data in the model file.  Somehow, that data was lost and now the models don't know where they belong.  On simple models like that they can be arranged manually by adjusting their location using the Movement Tool.  Just type in the values and they will lineup as they should.  The other option is to download the models and try again.

So the result of this is that I cannot duplicate the problem.  The models directly downloaded from Thingiverse properly merge.  How those models in your 3mf file lost their location data I have no idea.  But since the 3 pairs of models that were direct from Thingiverse did properly merge I have to say that I don't see a bug here.  I am inclined to close this.  Let us know what you think.
[image]

paulsartory, 2022-07-15:
Thanks Greg,

I have downloaded fresh copies of these files, extracted them with built in windows extract function. (I have used these files only as an example. The models I wished to merge are different and not amenable to manual merging)
They merge perfectly on all Cura 4 iterations and Arachne beta but not on (my) installations of Cura 5 and 5.1 beta. (They also merge in meshmixer and simply3D.) 

Somehow the location data must be lost when I load them into Cura 5. So whatever the problem is it is related to my installation of Cura 5. (I have done fresh installations of both)

So where do I go from here?

Regards Paul

paulsartory, 2022-07-15:
This how the appear prior to merging on Cura Arachne Beta:

[image]

And this is how they appear on Cura 5.1:

[image]

paulsartory, 2022-07-15:
Thanks for trying.

paulsartory, 2022-07-16:
I think I found a work around by merging in Arachne and exporting as a 3mf file then importing into 5.0. the models are accurately merged but still separate.

GregValiant, 2022-07-16:
It's an oddity for sure.  At least you are up and running.

paulsartory, 2022-10-11:
Thank you Greg,

I have attached a simple 2 STL cube that merges correctly in the Arachne Beta version but not in Cura 5.

paulsartory, 2022-10-11:
Greg,
I posted this problem on the Cura Ultimaker 3D Slicer User Group and the Cura users Facebook page but no one could help me. The only means of asking for help from Cura appeared to be the Bug Report option in the software itself.
Clearly that process is intended for "people like @fieldOfView (a main contributor to Cura and part of the "community" that maintains the project)" as you point out.
I am just a hobbyist that use your software an had a problem and wanted to resolve.
I was expecting a simple "how can we help" and to initiate an interchange  in order to solve the problem if possible.
In retrospect I should have used the Ultimaker Community Forum option on the Bug Report page

I apologise for using the wrong process but I do feel the responses to my mistake have critical of me rather helpful in solving the problem.
As for me I try to be helpful to anyone however "stupid" they may appear to be....its part of my nature

RegardsPaul Sartory

GregValiant, 2022-10-11:
I'm not an UltiMaker employee, or a code writer, or a contributor.  I started in on 3D printing to have something to do around the house during the Covid times.  So I'm a hobbyist like yourself but I have worked with gcode for a long time and so I look at the problems people bring here and try to figure out what's really going on in the hopes that it will be helpful to the Cura contributors and so that I can learn about 3D printing.  I do some of the basic legwork in the investigations so the developers don't have to (at least I hope so).
No investigation can take place without basic information like:  What's the model like?  Does it have any errors?  How was Cura set up?  What computer system and operating system?  What printer?, etc..  Without that information what can we do?  Your initial post didn't tell us anything except that you "had a problem merging and slicing" and you listed your printer as "none".  If you think about it, at that point all I can really say is "that's nice" because I don't know anything about the problem.
All these reports are taken seriously because nobody likes buggy software.  Major companies use Cura in mission critical projects and they can't have bugs getting in the way of their productivity.  If the problem can't be reproduced then it's impossible to point to a section of code and say "Aha!  There's where it's happening.".  So it is on the poster of the bug report to provide the information so an investigation can proceed.