Ultimaker/Cura #12591: Slicing Failed Error

cht685, 2022-06-23:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ender 400XL

### Reproduction steps

1. Added files
2. Changed the Print Settings
3. Clicked on "Slice"
4. Get annoying "Slicing failed" pop up

### Actual results

I receive this pop-up message:

Slicing failed
Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

### Expected results

Cura Slices the print and I start printing like every version prior to this one.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This is the latest issue I have on top of all the others I keep getting with Cura 5.0.0 now. Previous to this, if I opened a STL and tried to change the Print Settings after, it stops responding and goes full white until I Force Close it. I kept 4.13.1 as that is a stable release and 5.0.0 seems like it is still in BETA.

[cura.log](https://github.com/Ultimaker/Cura/files/8968668/cura.log)
[CURA 5_File That Causes Error.zip](https://github.com/Ultimaker/Cura/files/8968678/CURA.5_File.That.Causes.Error.zip)

[image]

GregValiant, 2022-06-23:
I moved the large model off the build plate and the small ones sliced.
I switched them around and the large model sliced.
With all the models on the build plate but the large model off to the side (rather than in the middle) it sliced.
Selecting all models and then rotating the group 45° allowed it to slice correctly.
Someone from the Cura Team will take a look at this.
(BTW...I didn't find any errors in the models (it appears you fixed them) but when a model slices in one orientation but not in another it has been my experience that there is a problem with the model.  That being said - if there is a problem with these models, it's very well hidden.

Split apart:
[image]

Rotated 45°:
[image]

GregValiant, 2022-07-05:
Using 5.1 Beta this sliced fine
There were a lot of these types of problems reported with 5.0.  I've been going back over the problem files and so far all of them have sliced with 5.1Beta.
I'll close this but if you have further comments you can post them and I'll be notified.

closed 2022-07-05 by GregValiant