Ultimaker/Cura #12592: Changes in machine setting > edit of starting g-code not recorded

ljohnsonnsb, 2022-06-23:
### Application Version

5.0.0

### Platform

Windows 10 

### Printer

Pulse ( a Prusa MK2 clone )

### Reproduction steps

I make a change to the starting g-code on a printer .

### Actual results

The changes are not saved or applied to the print.

### Expected results

The g-code to be included in the exported g-code

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I have 2 printers with similar names, (1) Pulse (2) Pulse Klipper

GregValiant, 2022-06-23:
This problem has been reported and I believe it will be fixed in the next release.  The duplicate is #12422.
As a workaround, after making changes to the either the StartUp or Ending gcode - click in the other text box.  That will set the changes and then you can close the dialog.  The bug occurs when the dialog is immediately closed without exiting the textbox.
Let us know if that works for you.

Joeydelarago, 2022-06-28:
Closing this, as it duplicates [#12422](https://github.com/Ultimaker/Cura/issues/12422)

This should be fixed in the 5.1 update.

Thanks

GregValiant, 2022-07-07:
@Joeydelarago 
I'm still getting this in 5.1 beta.  It is necessary to exit the textbox prior to hitting the close "X".  I've marked #12689  as a duplicate and I'll leave that one open.
#12689 is a duplicate of this and the poster was using 5.0.