Ultimaker/Cura #12594: Poor handling of thin walls resulting in additional travels and extrusions

jackbrick101, 2022-06-23:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Custom FFF

### Reproduction steps

Should be easy to reproduce; just slice a model with embossed font on the XZ or YZ plane and the issues should pop up.

### Actual results

The paths generated include little spurts of material to complete outer and inner wall paths. These spurts are so small and needless and only add more travel moves, retractions, and time to the print. Really increasing blobbing and stringing on the print.

This also breaks up the continuous outer wall extrusion line, so it has to pass back over itself to finish the path.

Disabling "print thin walls" somewhat addressing the issue, but not all of it.

### Expected results

The continuous outer wall should be printed continuously, not chopped up into bits. Same goes for the inner wall. I would expect the slicer to be able to identify what it would have to print after the continous line, and have the option to omit those small little spurts of material. 

If the wall to be printed can't be split in variable width lines, the ideal scenario would be to print it as a single line and stop extruding during the overlap (like the old "compensate wall overlaps" setting) or to omit that line.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

My guess is that the little blip is smaller than the "minimum wall line width", so instead of splitting it into two smaller line widths, it skips it as part of the continuous outer wall extrusion.
I tried reducing the "minimum wall line width", but even at ridiculously small numbers (that wouldn't print well anyway) it doesn't split to two.
I would hope disabling "print thin walls" would fix this, but it didn't - this behaviour still occurs. If "print thin walls" was a number we could turn up, maybe that would help address it.

At the end of the day, what I think should be addressed is the additional travel moves, retractions, and extrusions (which all lead to additional stringing, inconsistent small spurts of extrusion which result in blobs on the print) that occur after the continuous perimeter extrusion line is printed. I'd prefer the option to omit things that didn't fit within that continuous line rather than print them poorly afterwards.

In the project file, scrubbing through the layers anywhere the letters are printed, then going back and forth in the layer timeline, you can see the issue (extremely small spurts of material to 'complete' the outer or inner wall lines).

[ThinWalls_poor_behaviour_project.zip](https://github.com/Ultimaker/Cura/files/8969945/ThinWalls_poor_behaviour_project.zip)
[image]
[image]

GregValiant, 2022-06-23:
Thanks for the report.  There are two new settings in the Walls section:  "Split Middle Line Threshold" and "Add Middle Line Threshold".  Try setting them to 50% and then to 90% and see if it makes a difference for you.
There are situations where a single line width is all that will fit (like the downward point in the middle of the "y" and at the tops of the "r"'s) and so the toolpath can't do an out-and-back move while extruding.  As a consequence those tips are dead ends and can't be part of the continuous outer wall movement.  Adjusting those "Threshold" settings may make a difference in the toolpath.  (Whether or not it's a "good" difference you will have to see.)  I haven't played with those settings very much myself but they would seem to apply to your model.
Let us know whether changing those settings is a solution, no change, makes things worse, etc..

GregValiant, 2022-06-27:
I was tweaking one of my post-processors and happened to use your model.
(It looks odd because it's .2 layer height on the bottom, .1 in the middle and .3 on top. which was the post-process.)
I didn't do any print cleanup on this.  The overhangs were problems where I thought they would be problems.  This was printed without supports.
While this was printing I was watching closely and if I didn't learn anything else, I learned that I need a new layer cooling blower.
(Sorry about the poor image quality.  I need a new camera as well.)
[image]

jackbrick101, 2022-07-21:
Thanks for the response, Greg. 
That's similar to what I was running into on the "Greenery", sometimes the blobs (at the suspected issue areas, where a spurt is starting in the air) are a bit more pronounced but turning off "print thin walls" and lowering "Split Middle Line Threshold" helped a bit (but doesn't change the underlying slicing behaviour). 
It's just not ideal breaking up the outside path and extruding certain parts as little spurts, so overall I think the slicing could be handled slightly differently to produce cleaner and faster prints.

GregValiant, 2022-07-21:
FDM just doesn't handle printing over air very well.  The model dictates the path of the extrusion and I wouldn't want the software to "adlib" about how it should handle an overhang.  The overhanging parts of the letters really need support and they aren't going to get it.  Things like the center portion of the "e" have that straight line over air before the nozzle cuts back in.  It really needs something to set on but the support XY distance needs to be really small and you would still end up cleaning things up with a hobby knife because the support would probably stick to the model.
Now, all of that being said, if the letters had chamfers all around them then support wouldn't be needed and the letters would look good.
This was embossed in MS 3D Builder.  You can see that the letters are beveled.  One thing you might try in Cura is to enable "Make Overhangs Printable" in the Experimental section.  That might work well on your model.
[image]

jackbrick101, 2022-07-22:
Hey Greg, thanks for the response.

I understand the concept of printing in midair and support material. The point I'm trying to get across by starting this forum post is that there is a better way of handling the path generation in the slicing, because the current way in Cura 5 is subpar* (in my opinion, Cura 4.13 handled it better with it's "wall compensation"). I think Cura 5 has made great improvements with the switch to variable line widths, but has introduced new inefficiencies.
*subpar in how it deals with this specific scenario of a thin, outer surface section that doesn't get split into two thinner walls.

In the current behaviour, certain parts of the model start and stop in midair (those little "spurts" of material) rather than being part of a continuous extrusion path. Starting an extrusion path in midair will lead to worse results than having a continuous line dip into midair. Both have extrusions occuring in midair, which is unavoidable, but one just handles it better. 
Additionally, the current behaviour introduces extra travel and retraction moves (and moves the nozzle over the already printed outer wall) that are unneeded. Removing these would increase the speed and quality of the print.

I'm not a fan of the software adlibbing or guessing what it should do either, but there is none of that in the suggestions I've made. The two solutions I think would be helpful would be to give additional options to the user to control what dictates a "thin wall" (because currently, that IS adlibbed... there are thin walls that are still printed with "print thin walls" disabled) and to alter the slicing behaviour of thin sections (as described in my above post). 

I appreciate the suggestion to chamfer the model, but my goal is to get eyes on this issue in hopes of improving the program. This specific model isn't important, it's just where I noticed the new slicing behaviour.

GregValiant, 2022-07-22:
The Cura Team will certainly take a look.
From my point of view as a user and amateur coder, the new Arachne engine being rolled out at the same time as changing all the dialog boxes and controls to Qt6 was extremely ambitious.  There have been issues with both areas and it will take some time to dial everything in.  These types of helpful suggestions and bug reports are important and I'm sure are appreciated as well.

jackbrick101, 2022-07-22:
Thanks Greg, I totally understand (and agree - quite ambitious!). I appreciate you taking the time to go back and forth with me. 
Overall, I'm digging the variable line width in Cura 5, it solves a bunch of past problems. Looking forward to the next couple versions while everything gets dialed in!

GregValiant, 2022-07-22:
It's the "new thing" and although I didn't use Arachne beta's very much I went exclusively to 5.0 /5.1beta, and now 5.1.0.  It's different and yes, there will be some growing pains within the software as well as a learning curve as we users get used to it.