Ultimaker/Cura #12597: Multi material

Jesusday, 2022-06-23:
### Is your feature request related to a problem?

Cannot do MMU thangs with my MMU machines 

### Describe the solution you'd like

to do MMU thangs with Cura

### Describe alternatives you've considered

SuperSlicer

### Affected users and/or printers

MMU owners

Joeydelarago, 2022-06-28:
Hey, we will discuss this but some more information would be helpful. 

What features in particular do you think are important and what MMU are you using?

Thanks 🙂

Jesusday, 2022-06-28:
So I am actually just getting into multi materials my self so I could better answer that in probably a couple months but I think the biggest feature that could keep you competitive in this section would be this painting the colors on a stl feature but I have zero idea what it would require to make something like this work (maybe not worth it).https://m.youtube.com/watch?v=9Ztnec2k6SMThis video also goes over some features that might be needed/missing for a mmu setup?https://m.youtube.com/watch?v=PilDIaTaHHYI have not even messed with more than one tool head in cura yet, so I don't even know if you had an stl that was broken up into a couple peices, if you can print each section in different colors? I really have not came across any videos with cura using different colors (tool heads) so to be honest I don't even know what it can and cannot do at this point. Might be best to just make a video about it to highlight the features now that mmu is geting more popular (assuming there is not one already).Let me know if I can be of more help. Keep up the good work!Sean Sent from my Galaxy

Jesusday, 2022-06-28:
Oh and I am using the enraged rabbit carrot feeder (ERCF) that is kind of part of the Voron design but not really.Sent from my Galaxy

Joeydelarago, 2022-06-30:
Hey, Cura uses multiple extruders for multi-color printing and does not support these multi material feeders. We support up to 8 different extruders (8 colors).

I read up about the ERCF and it is very cool, the prints are stunning. Unfortunately this is not something that we are focusing on at the moment so I will have to defer this.

Pull requests are welcome!

closed 2023-07-14 by github-actions[bot]