Ultimaker/Cura #12602: Cura does not display as expected on a 4 monitor setup

yet-another-average-joe, 2022-06-25:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

not relevant

### Reproduction steps

Install 4 monitors ; mine are installed this way :

       [ T ]
[ L ] [ C ] [ R ]

Move Cura main window to monitor 3 or monitor 4

### Actual results

Cura does not display as expected on right and top monitors :

- window can be resized/maximized, but contents is not and Cura GUI does not reflect changes (view, controls, menus)

- sidebar GUI plugin controls are not updated if moved to monitors 3 and 4 (no caret, no text selection, nothing when typing, clicking, etc.)

### Expected results

Cura 5.0 should be usable on any monitor.

These screenshots show Cura maximized on right and top monitors (I had to reduce the JPEGs !)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[image]

Joeydelarago, 2022-06-28:
Hello, we will discuss this issue. 

Is it still possible to interact with the UI on your 3rd and 4th monitors even though they won't resize? And if so does the place you click match where the UI is?

Thanks for bringing this to our attention 🙂

yet-another-average-joe, 2022-06-28:
It is posiible to interact with somz UI elements.
Marketplace : as expected
Menus : they don't open, but blind clicking open the dialogs : they are there, but not visible
Model : nothing (selection, rotation, buttons don't work) ; but moving the Cura window to monitor 1 or 2, the model view is instantly updated (tested with view rotation).
Left button bar : moving the window to monitor 1 or 2 instantly makes the dialogs visible
Model : the same ; I can select it, zoom, but the state does not show ; it is updated when moving the window to monitor 1 or 2

(I suspect you want to know if the UI receives the mouse messages with the right coords ; I would answer yes... it seems the problem is related with the display only)

tlhintoq, 2022-06-28:
@yet-another-average-joe 
Mind if another multi-monitor user chimes in with a test?
Try setting the Windows scaling to the same value on all monitors.  Yeah I know its a pain in some cases: 4k on small built-in screen for example, so this is a just a test not a solution.
But I've seen many softwares are check the scaling on the *primary screen* not each individual screen, then try to apply that to all screens.  Then that fouls up all the interaction, calculations and so on.

yet-another-average-joe, 2022-06-28:
Scaling is 100% everywhere.

yet-another-average-joe, 2022-07-08:
I think that you can close the issue. It is most likely a configuration/driver problem. Today the problem disapeared.
My rig is an old but powerfull machine I built long ago to play Crysis full options, 3D Surround. It embeds 3x NVidia 780Ti + 1x Quadro K2000 (for CAD...).
When I installed Cura 5.0.0, the Quadro was removed, and I was using the last NVidia drivers with Kepler support (473.47). Two monitors were on the 3rd GTX (right + top).
Today, I DDU'd the NVidia drivers (473.47), reinstalled the Quadro, and installed the Quadro and GTX 378.66 drivers ; these drivers are old but known as super reliable with the Asus P9X79E-WS - this mobo has a very special circuit in order to support 4x PCIE 3.0 lanes instead of 2 or 3). Now my setup has one monitor per GPU, and the old drivers.

Cura now works as expected.

Sorry for the inconvenience, it's most likely a NVidia problem, not a Cura issue.

jldavis1969, 2022-07-31:
Please have a look at the following thread here on github:
Dual Monitor Error #12197
There is a fix at the bottom of the page that helped me.