Ultimaker/Cura #12603: Problem with the "tour de purge" on Dagoma Disco Ultimate Bi-Couleur

EddyCoubard, 2022-06-25:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Dagoma Disco Ultimate bi-couleur

### Reproduction steps

Hello,
This message to inform you that I think there is a bu using Cura 5.0.0 with a Dagoma DIsco Ultimate Bi-Couleur, if not, it is a bug of my mine ;).
All the words from Cura are in French, I guess it will be OK for you for your understanding (see the png files to see what I want to mean).
The problem is when using the 2 extruders with the "Double extrusion" in Cura, the "tour d'amorçage" does not not work as it should be with the Dagoma Disco Ultimate. Notice, it works perfectly with "Cura by Dagoma" from Dagona, but there is so many functionnalities in Cura that I would work with Cura.

### Actual results

1/ As you can see on Bug_Cura_2022-06-25_Dagoma_DU_Bi.3mf, the second extrudeur does not print on the "tour d'amorçage" on first layer. This is a problem, because there is not filament purge on this first layer.
2/ Afterwards, the filament purge is at each layer for extruder n°1, but it is one out of two for the extruder n°2. It is such a shame, because it is not easy to print as so with the extruder n°2 on the "tour de purge".
3/ On the file "Rond_Serviette-Céline_Bug_2022-06-25.3mf", the extruder n° 2 does not print till the layer 67. This is a shame, besause under, there is not filament printed, so it is impossible to print !
4/ There is a M109 in the middle of the G-Code that make the printer stpop in the part, and that is not good at all. I did not found how to relace this M109 by a M104 at this place.
You can see it on line 662 on Bug_Cura_2022-06-25_Dagoma_DU_Bi.gcode file, and line 47641 on Rond_Serviette-Céline_Bug_2022-06-25.gcode file.

### Expected results

What I could expect is to have with this kind of machine :
1/ For the first layer , a purge of the two extruders.
2/ Then, a purge of one extruder to one layer out of two (exemple 1st extruder on layer 3, 5, 7... and 2nd extruder on layer 2, 4, 6...), but exactly at the same place in XY on the "tour de purge".
3/ No M109 in the middle of the Gcode, only M104.

Thank you for reading, and guess I hace well explain the problem.
Best regards,

Eddy Coubard.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Bug_Cura_2022-06-25_Dagoma_DU_Bi.zip](https://github.com/Ultimaker/Cura/files/8984241/Bug_Cura_2022-06-25_Dagoma_DU_Bi.zip)

GregValiant, 2022-06-25:
@EddyCoubard I think I understand the problems.  Your printer has a very complex StartUp Gcode.  In addition, both extruders have their own StartUp and Ending codes.  There are also settings in the "Dual Extrusion" section that may be applied.  All-in-all it is a confusing combination with possible interference between the various gcode sections as they come together.  You are going to have to experiment a bit to get this dialed in.

-  _"T1 is not using the purge tower on the first layer."_
Enable the skirt (or brim) in "Bed Adhesion" settings.  Both extruders will take turns extruding it.  This will force a move to the purge tower.

- Your StartUp gcode appears to have incorrect information.  In particular this line appears twice:
G1 **E-.5** F5000 ;filament retract length
But your retract distance is 5.5 (not 0.5) and your retraction speed is 45mm/sec so I think both of those lines should be:
G1 E-{retraction_amount} F2700

- T1 is being called to start the print but when your StartUp Gcode ends the active extruder is T0.  When the print actually starts the "debut start T1" is in the gcode but T0 is active.  That is a conflict and it affects where the end of the filament is in relation to the nozzle.  Adding the line below as the very last line of your StartUp Gcode should fix that.
T{initial_extruder_nr}

- You are asking Cura to change printing temperature at the end of the first layer and so Cura puts in the M109 line.  To eliminate that line you should keep your "Initial Layer Print Temperature" equal to your "Print Temperature".  An alternative would be to go into the gcode file and manually change that line to M104 so there would be no waiting.

I don't see a bug here.  There are:
1. Conflicts within your Startup gcode.
2. The wrong tool is active at the end of the startup.
3. You have asked for a temperature change and you are getting one.
4. Allowing Cura to put down a skirt (4 loops I think would be sufficient) will force both extruders to use the purge tower on Layer:0.

I will leave the bug label for now.  Let us know if making those changes works.

EddyCoubard, 2022-06-26:
Hello Greg,

Firstable, I have to thank you for answering so quickly on a saturday, it is very nice of you.

I have made some tests thanks to your tips, and see below the results. All the files in Bug_Cura_2022-06-25_Dagoma_DU_Bi_V-02.zip.
On the two examples, I have changed the color of the first extruder in black to be more noticeable.
A video made for you may be for more understanding here: https://we.tl/t-c1M1atltLF (I do not know if I can send such as this kind of video on the Github).

- T1 and Purge at first layer : See on file Bug_Cura_2022-06-25_Dagoma_DU_Bi_V-02.3mf.
I had not understood why it was necessary to activate the skirt, but thanks to you, I understand it quite well now. My misunderstood had made me mistake on the sequency of the extruders at the start, I think it is OK from now. The result is that the skirt is necessary in that way, not so elegant in my mind, but it does the job ;).

- Retract at Start-Up Gcode.
An old test of mine that should have been changed. Thank you to have seen it, even if it was not the subject.

- Temperature and M109.
The different temperature at first layer is a need for my printer (no hot bed), and with a M109, the waiting for decreasing the temperature is in the part, and this in no good.
You can see it on the video at 1 minute 05 seconds.
As so, should not it be better to be done on the purge tower?
I change it manually with a M104, but I would rather have an option somewhere in Cura to have the choice to put a M109 or a M104, just not to forget to change it in the GCode as I often do actually.

- All theese points seems to be OK for me from now, but there is one that is not solved yet for me, you can see on Rond_Serviette-Céline_Bug_2022-06-25_V-02.3mf.
With the presious tips, the purge tower is OK on layer 1.
But from layer 2 to layer 66, there is no filament (the pink one) in the tower purge. And at layer 67, the filament from the second extruder has no support to be printed on the tower purge! (you can see it on picture Layer_01.png, Layer_66.png and Layer_67.png).
And after, for the extruder n°2, the print on the tower purge is on 69, 71, 73 and so on. There is no print on the tower purge on layer 70, 72, 74 and so on... I think it is a mistake.

I understand that as there is no need to use extruder n°2 till layer 67, it is not useful to print with the extruder n°2 in the purge tower, but this could work if the position of the purge tower is the same for extruder n°1 and n°2.

Did I make somethong wrong in that case?
What could be solution, without doing what I have done, printing another part unusefull as you can see on picture Additional_Part.png?

-----------------------------------------

Outside of my problem left below, I have let me imagine what could be the sequency in the way where the extruder n°1 and n°2 printing at the same place in the tower purge:

Specific for first layer
   - Layer 1:
      - Skirt extruder n°2
      - Part extruder n°2
      - Tower Purge extruder n°1
      - Part extruder n°1

Then for standard layer with two colors:
   - Layer 2:
       - Part extruder n°1
       - Tower Purge extruder n°2
       - Part extruder n°2
    - Layer 3:
       - Part extruder n°2
       - Tower Purge extruder n°1
       - Part extruder n°1 
   - Layer 4:
       - Part extruder n°1
       - Tower Purge extruder n°2
       - Part extruder n°2
    - Layer 5:
       - Part extruder n°2
       - Tower Purge extruder n°1
       - Part extruder n°1 
and so on...

If only one color till the 2 colors reach:
   - Layer 2:
       - Tower Purge extruder n°1
       - Part extruder n°1
   - Layer 3:
       - Part extruder n°1
       - Tower Purge extruder n°1
   - Layer 4:
       - Tower Purge extruder n°1
       - Part extruder n°1
   - Layer 5:
       - Part extruder n°1
       - Tower Purge extruder n°1
and so on... We can imagine in that way that the purge tower could be with less volume during this specific printing wity only one color...

And what about having the tower purge inside of the part in the internal filling? It should very interesting, and using less filament. What do you think about it?

-----------------------------------------

I have got another question: how to make in Cura the automatic saving of a file when leaving Cura (Alt + F4), as all the standard softwares from now do. I have lost so many time because no saving at Cura leaves. I have not found any option on it. Thank you to answer also to this little question.

Best Regards,

Eddy Coubard
[Bug_Cura_2022-06-25_Dagoma_DU_Bi_V-02.zip](https://github.com/Ultimaker/Cura/files/8986411/Bug_Cura_2022-06-25_Dagoma_DU_Bi_V-02.zip)
.

GregValiant, 2022-06-27:
When I sliced a model that didn't use T1 for extended amount of layers it showed the same problem that you noted.  When T1 finally becomes active it is printing over air.  So long as it wipes on the tower this should work but I would imagine it looks sloppy.  You may have to make the tower larger in diameter to provide enough volume for all the spaghetti.  A fix for that might be to print the T1 portions of the tower using T0 as it is already there and active.  The tower would build and there would be material there for T1 to print on.  That might not work if the materials are dis-similar.
So I think you have brought up 3 issues.
1. The M109 lines that occur mid-print effectively pause printing with the nozzle in contact with the print.  M109 can cause a 10 second delay even when the temperature is already at the set point.
2. Gaps in the purge tower when there are layers that don't involve one of the extruders.  When the tool switch is made the newly activated extruder prints over air at the purge tower.
3. A true auto-save function that would save a project in the background and then prompt on closing.  (That might be considered a new feature).

EddyCoubard, 2022-06-29:
Hello Greg,

Just have a look on Rond_Serviette-Céline_Bug_2022-06-29_V-02 file !
I have just exchanged the two extruders, and it seems to work fine in the tower bridge;) .
I just do not understand why the extrudeur n°2 prints one layer out of two in the tower bridge, and the extruder one prints all of them? If you could explain it to me...
I would try to make some more tests till the end of the week to better understand how Cura works with the two extruders...

Best Regards,

Eddy Coubard.

[Rond_Serviette-Céline_Bug_2022-06-29_V-02.zip](https://github.com/Ultimaker/Cura/files/9014761/Rond_Serviette-Celine_Bug_2022-06-29_V-02.zip)

GregValiant, 2022-06-30:
Hello @EddyCoubard.
I have "virtual" multi-extruder Creality printers.  I can change any of their printer models to use up to 4 extruders.  I do that in order to understand the problems that people might have.  In addition I have spent some time playing with the Geeetech multiple extruder machines as well as the Ultimaker multi-extruder machines, and some IDEX printers.  That has given me some knowledge of the problems that people might have with the multi-extruders with multiple hot ends as well as the "multiple-in-one-out" style printers.
Unfortunately, it isn't the same as actually owning a real one and printing on it.  I have managed to build up some theoretical knowledge but I lack practical knowledge of the printers.  So I am limited in the amount of help I can provide.
Experimenting with the machine and doing things like switching which extruder starts a print is a necessary evil on these little robots.  Dialing in a StartUp Gcode that works for you is another part of that.
I'm sure the Cura Team will take a look at this when things quiet down from the release of 5.0.  In particular I can see where the M109 commands are a problem.  A "multiple-in-one-out" printer should not have mid-print M109 lines.  Those type hot ends are a fairly recent development and it will take some time for the software engineers to incorporate the different requirements of that hardware into Cura.