Ultimaker/Cura #12604: Cura 5.0 keeps on crashing with OSx 12.4

Beetleclub, 2022-06-25:
### Application Version

5.0.0

### Platform

MacOSx 12.4

### Printer

Ender 5

### Reproduction steps

1. while editing printing profiles
2. while reviewing slicing results

### Actual results

Cura crashes

### Expected results

no crash

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8984575/cura.log)

nallath, 2022-06-28:
Cura 5.0.0 requires at least MacOSx 12.5. See https://support.ultimaker.com/hc/en-us/articles/360011889579-On-what-operating-systems-is-Ultimaker-Cura-available-

Beetleclub, 2022-06-28:
That‘s not correct. Ist requires 10.5 or above. Which should be so with 12.4. Just as a hint, 12.5 is available only as Beta version. So the problem is still there. Also 4.13 chrashes with Monterey, but less frequent.

fieldOfView, 2022-06-28:
To clarify:

There's a macOS 10.12 ("Sierra") and a macOS 12 ("Monterey"). Cura 5 is not compatible with macOS 10.12; it requires macOS 10.15 ("Catalina") or newer.

Beetleclub, 2022-06-28:
My mistake. The „x“ in the  initial post was misleading. I am talking about Monterey MacOS 12.4. Hope this Males ist More clear.