Ultimaker/Cura #12607: Unable to remove unused default materials from preference/materials

RolandLarocque, 2022-06-26:
### Application Version

5.0.0

### Platform

MacOS

### Printer

Creality Eder 5-plus

### Reproduction steps

Go to prefrences/materials and try to remove TiZYX Flex   
Unable to delete

### Actual results

Unable to delete

### Expected results

The materials you don't use , you should be able to remove them

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No files required

GregValiant, 2022-06-26:
No you can't.  The chances of me ever using GooFoo HIPS is exactly zero.
I've tried deleting the files from the Materials folder and that didn't work.
The "creality_base.def.json" file has a section for
"exclude_materials": [
            "Vertex_Delta_ABS",
So far I haven't had any luck putting new ones into the list.
@RolandLarocque, Not being able to remove filament brands is "Expected Behavior" and so I'm leaning toward changing this to a "Feature Request".  A user would be allowed to remove materials except for the "Generic" materials which would continue to be the base for any Custom materials.

Let me know what you think about that.

Joeydelarago, 2022-06-29:
Hi we will discuss this as a feature request.

However if you would like to delete these materials right now. It is possible if you have access to the installation folder of Cura. 

We are both on MacOS so this will be easy to explain. 

1. In file explorer go to Applications. 
2. Right click on Cura and click the "Show Package Contents" option.
3. Navigate to Contents/Resources/share/cura/resources/materials
4. You can delete the material files here

Ghostkeeper, 2022-07-08:
You can't delete those files on MacOS since the notarising of the package will then complain that the package has been modified and is potentially security-compromised. On Linux you can't delete them either without significant knowledge (since the AppImage is a read-only image). On Windows you could delete them from Program Files.
However in all of these cases, messing with the shipped materials is not advisable. It could easily cause Cura to crash if you don't know which profiles the code depends on (e.g. a lot of printers have certain materials as defaults). And we will never fix those crashes nor provide workarounds, since you're essentially modifying the source here. You'd be on your own.

We decided to reject the feature to allow built-in materials to be deleted from the GUI as well. Cura itself is not able to modify the files in the installation itself too (e.g. modifying files in Program Files requires admin access). We could implement a feature to "hide" them, but that necessarily complicates the workflow for users, e.g. if they forgot that a material could also be hidden by someone (or even by themselves) half a year ago.

If your aim with removing them is to improve performance, consider using the [Startup Optimizer](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/StartOptimiser) plug-in. If your aim is to reduce clutter of the interface, your only option is to mod the resources folder (as described above, or by adding `exclude_materials` to the metadata of the printer definition).

closed 2022-07-08 by Ghostkeeper

dtrigger0, 2023-03-05:
A tad off topic, but for Windows 10, the files are located in:
     C:\Program Files\Ultimaker Cura #.#.#\share\cura\resources\materials.  

Simply deleting them worked just fine for me.  
BTW: #.#.# is of course the Cura version number being used.  

Also, if you get "File Access Denied", and your user account has admin rights, then just click continue, and you admin rights will be applied.

carlosjln, 2024-10-24:
What a PITA 😩 a feature that quite a lot of user do not want and yet is forced to clutter the UI/UX. This could have been an "Ultimaker Materials Plugin" where users could pick and choose which materials they wanted from the library.

After all, if now there's a plugin able to hide them, then there's a way and not an excuse that the app can't delete files due to permissions, simply change the approach.

fieldOfView, 2024-10-25:
The mentioned Startup Optimiser plugin will let you remove all non-generic, non-user-generated materials from Cura, whether you want to do that to reduce clutter or to speed up loading. So there's a solution; it is up to you to apply it.