Ultimaker/Cura #12608: inexplicable high speed values in gcode

exec79, 2022-06-26:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

custom (voron)

### Reproduction steps

imported settings profile from cura 4.6.1
slicing any modell with that settings shows the problem

### Actual results

gcode contains very high speed values like:
G1 F110769.2 X157.194 Y146.225 E0.00124
G1 F720000 X157.126 Y146.22 E0.00002
G1 F7200 X156.556 Y146.203 E-0.00019

### Expected results

speed values i set.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

doublechecked my speed values.
using built in settings profiles dont show the problem.
[image]
[CFFFP_bringa22.zip](https://github.com/Ultimaker/Cura/files/8986439/CFFFP_bringa22.zip)

GregValiant, 2022-06-27:
Thank you for the report.
Your images show that you have a post-processor active but the project file does not.  What post-processor was it?
I am not getting any odd speed numbers when I slice the project.
M204 S5000
G1 F600 Z0.4
G0 F21000 X153.744 Y153.122 Z0.4
G0 X153.305 Y154.474
;TYPE:SKIRT
G1 F600 Z0.2
G1 F2700 E1
G1 F2400 X153.744 Y153.122 E0.04587
Your Initial Layer Travel speed is 350 and that's what it is in the gcode file I generated.  Z-hops are at 10mm/sec so that's on the money.  Your retract speed is 45mm/sec and the first extrusion goes down at 40mm/sec.

I am unable to duplicate the problem but as I noted, the project file doesn't have any post-processors active and so we may be talking apples and oranges.  Maybe you could post another project file with that post-processor active?  I will leave this as a bug report for now.

exec79, 2022-06-27:
Thank you for your reply!
if i remove the post-processor script (which was a 3rd party script that checks the min. and max. positions of the actual print, writes in the gcode and a klipper script reads it, then probes only the required printing area for mesh leveling)  the problem still there. As i said, if i build up the profile from scratch it works as it should. i also attached the generated gcode, you can see some of these strange speed values for example from line 45786 
[image]

[CFFFP_bringa22.zip](https://github.com/Ultimaker/Cura/files/8996408/CFFFP_bringa22.zip)

GregValiant, 2022-06-28:
I did a search and these showed up in a different area.
G1 F43636.4 X157.772 Y146.28 E0.00315
G1 F110769.2 X157.194 Y146.225 E0.00124
G1 F720000 X157.126 Y146.22 E0.00002
G1 F7200 X156.556 Y146.203 E-0.00019
G1 X155.986 Y146.186 E-0.00075
G1 X155.035 Y146.214 E-0.00086
G1 F169411.8 X154.484 Y146.258 E0.00077
Along with the "F" values the "E" values are wonky as well.  There are three negative extrusions in a row.  I am getting the same results in the same lines when I switched to Absolute extrusion.
G1 F43636.4 X157.772 Y146.28 E2138.82491
G1 F110769.2 X157.194 Y146.225 E2138.82616
G1 F720000 X157.126 Y146.22 E2138.82618
G1 F7200 X156.556 Y146.203 E2138.82599
G1 X155.986 Y146.186 E2138.82524
G1 X155.035 Y146.214 E2138.82438
G1 F169411.8 X154.484 Y146.258 E2138.82515
I can say that after rotating the models on the build plate the problem seemed to resolve itself.
Maybe @Ghostkeeper can find some time to look at this.

GregValiant, 2022-06-28:
- Grouping the models and then rotating the group on the build plate seems to make the problem go away.
- Right at the problem point there are some very short line segments.  Changing the "maximum resolution" to 0.6 appears to fix the issue.

closed 2023-10-05 by github-actions[bot]