Ultimaker/Cura #12611: unexpected error during slicing

pflonsch, 2022-06-26:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender-3 Pro

### Reproduction steps

1. Loaded file "LVL2_LibraryRightA.stp"
2. Clicked "Slice" with print settings I used several times before
3. Tried to rotate object
4. Tried to restart Cura
[image]

### Actual results

Cura stopped slicing with an Error message.

### Expected results

Cura should slice, or give an error message with a hint to what could be the problem.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[LVL2_LibraryRightA.zip](https://github.com/Ultimaker/Cura/files/8987575/LVL2_LibraryRightA.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8987577/cura.log)

GregValiant, 2022-06-27:
I made no changes to your 3mf file and it sliced fine for me.  Then I moved it on the build plate and it wouldn't slice.
There have been a number of reports lately on this sort of behavior with 5.0 and it's been for different reasons.  This does not appear to be a model problem.  Sometimes it slices and sometimes it hangs.  As a workaround - try moving it here and there on the build plate, change the "Initial Layer Height" up or down by .01mm, rotate it a little bit, that sort of thing.
This was the first attempt.  I was feeling pretty good there for about a minute.
[image]

pflonsch, 2022-06-27:
Thank you for your support!
I tried rotating the whole part by 90°, which did not work. Now I just shifted it 1 mm to the side, and it slices.
Thanks again!

GregValiant, 2022-07-05:
I think I can close this one now.  It kind-of sliced in 5.0 and I had no trouble at all with 5.1 Beta.
@pflonsch if you have further thoughts on this problem then post your comments here and I'll get an email.

closed 2022-07-05 by GregValiant