Ultimaker/Cura #12612: Strange non-existing moves on preview

carribeiro, 2022-06-27:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. Loaded two files to print "one at a time" (on purpose; better quality, slightly faster overall).
2. Sliced with the parameters saved in the 3MF file.

### Actual results

1. On preview, after the last layer of the model on the right side of the printer is finished, there's a lot of strange lines going to the origin, The lines are of several types. 

Image as seen on preview.
[image]

2. Checking the GCODE file, the lines are not there. So it seems that preview is rendering some information that is only present in memory.

[image]

### Expected results

Preview should not show these strange moves after the part is finished. The file seems to be ok (I'm about to test it, bt reported first as not to forget).

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The ZIP file contains the STL models, the 3MF file, LOGFILE, and the GCODE as generated.

[Custom 3D Frame.zip](https://github.com/Ultimaker/Cura/files/8987890/Custom.3D.Frame.zip)

carribeiro, 2022-06-27:
There's another problem, but it seems to be an UNRELATED problem. I loaded the GCODE file on Cura and it's not honoring the "one at a time setting". It's really strange.because the GCODE file is correct; seems the GCODE rendering gets broken because of the same layer number happening more than once.

closed 2022-06-27 by GregValiant

GregValiant, 2022-06-28:
5.0 was a big change.  All the dialog boxes had be be revised, Arachne was folded in.  Together those things have generated a lot of bug reports.  I think it's better to have a duplicate than to have something go unreported so it certainly wasn't a problem.
BTW I was looking at another "one at a time" problem earlier today and your issue popped up on my computer.  There they were - stupid lines to the origin.