Ultimaker/Cura #12614: unable to reach the ultimaker account server

BretG137, 2022-06-27:
### Application Version

5.0.0

### Platform

debian 12

### Printer

mini delta

### Reproduction steps

Load cura with:
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-linux.AppImage

### Actual results

cura loads fine but says:

Log-in failed
unable to reach the ultimaker account server

### Expected results

login to ultimaker account works fine on 4.13.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Log-in failed
unable to reach the ultimaker account server

BretG137, 2022-07-20:
It looks like it some problem with gnome keyring
`2022-07-19 23:44:55,581 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]: Exception: No keyring backend present
2022-07-19 23:44:55,582 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]: Traceback (most recent call last):
2022-07-19 23:44:55,582 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]:   File "cura/OAuth2/KeyringAttribute.py", line 68, in __set__
2022-07-19 23:44:55,582 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]:     keyring.set_password("cura", self._keyring_name, value)
2022-07-19 23:44:55,582 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]:   File "keyring/core.py", line 60, in set_password
2022-07-19 23:44:55,583 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]:   File "keyring/backends/fail.py", line 25, in get_password
2022-07-19 23:44:55,583 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]: keyring.errors.NoKeyringError: No recommended backend was available. Install a recommended 3rd party backend package; or, install the keyrings.alt package if you want to use the non-recommended backends. See https://pypi.org/project/keyring for details.
2022-07-19 23:44:55,583 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]: Exception: No keyring backend present
2022-07-19 23:44:55,583 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]: Traceback (most recent call last):
2022-07-19 23:44:55,583 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]:   File "cura/OAuth2/KeyringAttribute.py", line 68, in __set__
2022-07-19 23:44:55,584 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]:     keyring.set_password("cura", self._keyring_name, value)
2022-07-19 23:44:55,584 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]:   File "keyring/core.py", line 60, in set_password
2022-07-19 23:44:55,584 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]:   File "keyring/backends/fail.py", line 25, in get_password
2022-07-19 23:44:55,584 - WARNING - [MainThread] cura.OAuth2.KeyringAttribute.__set__ [78]: keyring.errors.NoKeyringError: No recommended backend was available. Install a recommended 3rd party backend package; or, install the keyrings.alt package if you want to use the non-recommended backends. See https://pypi.org/project/keyring for details.`

BretG137, 2022-07-20:
There is a further problem with 5.1.0:
`./Ultimaker-Cura-5.1.0-linux-modern.AppImage 
[59833] Error loading Python lib '/tmp/.mount_UltimaV7qNkQ/libpython3.10.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /tmp/.mount_UltimaV7qNkQ/libpython3.10.so.1.0)
`
If I try adding the ld_preload i get:
`bash -c "QT_QPA_PLATFORM=xcb LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.1.0-linux-modern.AppImage"
/tmp/.mount_Ultima2mEwly/Ultimaker-Cura: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.35' not found (required by /tmp/.mount_Ultima2mEwly/libgcc_s.so.1)
/tmp/.mount_Ultima2mEwly/Ultimaker-Cura: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/.mount_Ultima2mEwly/libgcc_s.so.1)
`

nallath, 2022-07-20:
Did you also try the other version of the appimage? We now provide two.

BagelOrb, 2022-08-12:
+1
I get the same error message when I try to open the app image on my Ubuntu 22.04 boot.

BagelOrb, 2022-08-12:
Downloading and running the `Ultimaker-Cura-5.1.0-linux-modern.AppImage` instead of the `Ultimaker-Cura-5.1.0-linux.AppImage` solved that last error for me.

closed 2026-08-16 by github-actions[bot]