Ultimaker/Cura #12615: Mirror and select face to align to build plate tool.

ikeji, 2022-06-27:
### Application Version

5.0.0

### Platform

Linux

### Printer

Ender3

### Reproduction steps

1. Load any object (in this case calibration cube: https://www.thingiverse.com/thing:1278865 )
[image]
2. Use mirror tool, swap left and right. (Red allow)
[image]
3. Use "select face to align to build plate" tool, click that surface that have "Z".

### Actual results

Cura rotates the "Z" surface (clicked surface) to top. (In this case,"Nothing changed")
[image]

### Expected results

Cura rotates the "Z" surface (clicked surface) to bottom (touch to build plate)
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

If I click "X" surface, cura moved the surface to top too.
[image]

The "select face to align to build plate" tool without Mirror tool works as intended.

I guess the mirror tool confused the "select face to align to build plate" tool.

Joeydelarago, 2022-06-29:
Hello, this one has come up before. Not sure how it sneaked back in. We will discuss this issue.

Thanks for informing us

Ghostkeeper, 2022-07-13:
For me this issue doesn't happen. After mirroring, clicking on a surface still moves the surface facing down to the build plate.

We are thinking that this might be different per computer. The "lay flat on surface" function makes a render of the scene's normal vectors and rotates the object based on the normal vector of the pixel you click on. Maybe your GPU renders the normal vectors differently and that's what's causing the issue.

Because it's hard to reproduce and seems to be rare in the field, we've decided to defer this issue for now. If it starts becoming more frequent we can reconsider the priority.

UEPONYA, 2022-07-14:
I can reproduce.

1-axis inversion causes the selection surface to face up
With 2-axis inversion, the selected surface faces down

CPU: Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz (3408 MHz)
Memory: 63593 MB
Concurrency: 8
OS Version: Microsoft Windows 10 64-bit (Build 19044.1766)
Graphics Card Vendor: NVIDIA Corporation
Graphics Card: NVIDIA GeForce GTX 1060 6GB/PCIe/SSE2
Graphics Card Memory: 6144 MB

ikeji, 2022-07-14:
I tested on two machines and the problem is still happen.
Machine 1: Ryzen 4750G with radeon integrated GPU, Ubuntu 20.04
Machine 2: Intel Core i5-8350U with Iris integrated GPU, Windows 10.

3Duke3, 2022-07-15:
I can also reproduce this issue.

Intel i7-8700k 
Nvidia RTX2080
Cura 5.0