Ultimaker/Cura #12619: irioning inconsistent since 4.13

jxer, 2022-06-27:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

ender3 v2

### Reproduction steps

Has something changed in the ironing algorithm?

I NEVER had trouble with ironing in 4.x versions. I am now getting random under extrusion during ironing since moving to 5.x

### Actual results

under extrusion in random sections of ironing

### Expected results

consistent ironing
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothin

Joeydelarago, 2022-06-29:
Could you attach your project file so we can see your settings?

jxer, 2022-06-29:
is there a way to send it privately. this is an actively sold item

Joeydelarago, 2022-07-06:
If you could open the project file where you are having the issue (So that the settings are loaded), delete your model and then replace it with something like a calibration cube. This would get the information we need to us.