Ultimaker/Cura #12622: Some settings are not available in "Per Model Settings"

M-ABCDEFG, 2022-06-28:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ender3

### Reproduction steps

A segment within within the model shall be printed with support, while the rest with "Spiralize Outer Contour" option enabled (the option of the vase mode).

A "Support Blocker" plugin was used to enable or disable the vase mode for different segments.

### Actual results

Different configurations of support blockers were tried. 
Nothing changed.

### Expected results

Expected to find "Spiralize Outer Contour" option available under the "Normal model" "Support Blocker".

"Support Blocker">"Normal model">press "Select settings"> there is no "Spiralize Outer Contour" 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

GregValiant, 2022-06-28:
Thanks for the report.
Spiralize creates a single spiral (a single extrusion) all the way up a print.  If the nozzle has to leave the spiral for any reason then spiralize fails because it has been interrupted.  By definition, anything that requires supports can't be sprialized because the nozzle has to leave the model to create the support.  This isn't the only setting that is not available in Per Model settings.  There are numerous settings whose "settable_per_mesh" property is "false".  Each has a reason why that is so.
I'll leave the bug tag on for now so you can reply, but what I see here is "expected behavior".  Spiralize just doesn't work correctly if the spiral motion is interrupted for any reason.

M-ABCDEFG, 2022-06-28:
[image]

Thanks for giving me the chance to comment.

That is the model, what I wanted is to build a vase (which is hollow) **upon** a solid base, using the spiralize option (the vase mode) because it is much faster and  almost no apparent seams.

To do so a "support blocker" was added to "segment A" to make it solid, as the rest of the model "Segment B" would be printed spiralized.

Maybe that shall be put into new features.

So, is it possible to guide me where or how would the source code be opened to change "settable_per_mesh" property? 
It would be a good idea to try the slicing differences.

Goodbye

GregValiant, 2022-06-28:
_"...how would the source code be opened to change "settable_per_mesh" property"_  I don't know.  I tried to do it with a setting once (just to see if I could) and I couldn't get Cura to respond to it.
We don't know what the effect would be on the slicing engine trying to do that either.  In the case of "spiralize" it probably wouldn't be good.  The settings that have been defined as not "Settable per mesh" were no doubt done so for a reason.  You will notice that there are no "Travel" settings (at all) that can be configured "per mesh".
Ignoring Z-Hops which are a special case - the Cura toolpaths keep going up and up.  Put down a layer at .2, move up, put down a layer at .4, etc.
But spiralize is different.  The Z is constantly moving up all through the course of the layer until the nozzle gets back to the start where it is finally 1 layer height above the previous layer.  It's close to a true spiral motion.  If the nozzle were to jump out of the spiral to do support, or an adjacent feature, or another model, then the spiral is broken and there would be a seam.  When the nozzle comes back would it need to drop from say .8 to .657 in order to continue the spiral?  I'm thinking at least a blob if not a crash.
When I run into a situation like you have there I create two gcode files and splice them together.  There might be a solid base with a spiralized feature as a topper.  The XYZE need to be synced between the base and the spiral part but creating the transition code is more tedious than difficult.
Looking at your model I don't think either section can be spiralized.  It is called "vase mode" for a reason.  That upper section seems to be made of separate finger looking things sticking up.  Those cannot be spiralized.  The lower portion is hollow and so it cannot be spiralized.
A model suitable for spiralize must be solid through and through.  Cura can only encounter 1 outside wall all around the part.  In the case of your base, there is an outside wall facing outward with air on one side and model on the other, and another outside wall facing inward that has air on one side and model on the other.  Can't do that in spiralize because the nozzle has to leave one wall, go to the other, and then come back to the first wall.  That will leave a Z-seam on both of the walls.
If you want to post the models I'll take a look but from what I see in your images, spiralize simply won't work on those models.

M-ABCDEFG, 2022-06-29:
Thanks so much.
In my case, the vase mode -and without the extruded leaving the model to print a support- the extruded continued to print an inner wall. That is because there is a shape within the circumference of the model, and the inner shape is close enough to the outer shape (that of the skin), and which made both the inner and the outer shapes fuse, making the printed part more consistent.
That is the my case regarding the **upper part of the model (segment B)**.

Regarding the **lower part of the model (segment A)** (the part with the fingers like things extruded outwards), according to the required model it should have been solid.

So the issue is about a way to build some layers, sort of a platform, **below** the vase, before printing the vase.
So its base be think like the next 2 vases but not  too thin like the 3rd one
[image]
[image]

[image]

This would help not only in vases as you may notice. Actually what was required was to print swiftly and smoothly a screw with thick solid head, and it shall be printed vertically not to affect the smoothness of the screw. 

Like this one, as you may notice the screw head is thick.
[image]

Joeydelarago, 2022-06-30:
Hello, unfortunately implementing a combination of vase mode and non vase mode would be very complex. I think that this would also not solve your problem. If printed in vase mode, the screw would be extremely weak since it would only be one extruded line thick. 

I have two suggestions for you.

First if you are having problems with fitting your screw into a thread, read through [this post](https://www.reddit.com/r/3Dprinting/comments/pze447/great_cura_feature_more_should_know_about_slicing/) about slicing tolerance.

If you would like to create a hollow screw I suggest doing a boolean subtract with a cylinder in the centre of your screw. I'm not sure how familiar you are with CAD software but this is very doable without much knowledge using TinkerCAD (an in browser CAD software). 

Good luck with your project!