Ultimaker/Cura #12623: Wall count different on overlap modifier settings depending on print sequence mode

mutrised, 2022-06-28:
### Application Version

5.0.0

### Platform

windows 10

### Printer

Ender 3 Pro (same for Wanhao D12, CR-10...)

### Reproduction steps

1. Have a piece to slice
2. Add a mesh to change settings on specific parts of the piece to print
3. Set the mesh as modify settings on overlap, infill mesh only
4. Base settings is 2 walls
5. Add 2 extra walls for the overlap to have 4 wall on the part that requires more strengh
6. Select One a a time as print sequence setting

### Actual results

The part get 4 wall added (1 out, 1 inn, 1 out and 1 inn) when print sequence is set to one at a time
So 6 wall instead of 4, with odd extra outer wall in the middle.
[image]

### Expected results

The same as when print sequence is set to all at once, have 2 walls added to reach 4.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[wall_count_issue_print_oneatatime.zip](https://github.com/Ultimaker/Cura/files/9000773/wall_count_issue_print_oneatatime.zip)

GregValiant, 2022-06-28:
I saw your post on the UM Forum.  This was easy to duplicate.
Thanks for posting it here.

mutrised, 2022-06-28:
Thanks for being so reactive ;)

mutrised, 2023-04-14:
Since there is no news on this for the moment, for those who would be interested in a workaround, that's not a proper one but it fit my needs at least: 
the **Extra infill wall count** can be used.
Walls are added the same way, the only difference is these are infill instead of inner walls,
 but it prints faster than outer walls so I'm ok with it.