Ultimaker/Cura #12625: Incorrect skirt placement - Cura 5.0.0

Fabbro03, 2022-06-28:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Anycubic Mega Zero

### Reproduction steps

1. duplicated two objects and separated them
2. sliced

### Actual results

The skirt goes through the objects.

[image]

### Expected results

A correct skirt placement.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file: [AMZ_supp_tubo-telaio.zip](https://github.com/Ultimaker/Cura/files/9001316/AMZ_supp_tubo-telaio.zip)

smartavionics, 2022-06-28:
The objects need to be placed flat on the buildplate. As you have it, they are at an angle and so the skirt goes underneath.

After adjusting their orientation...

[image]

Fabbro03, 2022-06-28:
I apologize infinitely, you are right. I tried it and it worked. But there is one thing I don't understand in the rotation without snap.
The piece has the face at a 95-degree angle. So after importing it, I rotate it without snap by 95 degrees. Now to make it flat, I have to either press "lay flat" or I have to re-rotate it, without leaving the left button, at any angle and bring it back to 0°, then it lays it flat. So why doesn't it lay it flat right away, since the second rotation rotate it by 0°?

Fabbro03, 2022-06-28:
I've just added the pull request (https://github.com/Ultimaker/Uranium/pull/835) to show the decimals of rotation angle, so it can prevent the incorrect rotation of the object that can occur with only the integer display of the angle.

Can it be integrated within the next version of Cura?

GregValiant, 2022-06-30:
@Fabbro03 I'm going to close the bug report since it appears to have been resolved.  If you feel that is in error let me know here.
As to whether your PR can be implemented for the next version - that would be a question for the Cura Team and the place to ask that would be on the PR page.

closed 2022-06-30 by GregValiant