Ultimaker/Cura #12628: Slicing Failed

cpacker001, 2022-06-28:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Anycubic Mega Zero

### Reproduction steps

1. Open file in Ultimaker Cure 5.0.0
2. Clicked on Slice
4. Slicing started
5. Then got Slicing Failed.

### Actual results

After getting Slicing Failed it gave me a report bug button.
Screen went back to the beginning of the slice process.

### Expected results

It should have sliced the model and created a gcode file
[image]
[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/9003170/cura.log)
.
Cannot add project File because it is too big 32MB.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Tried to load file but it is too big.

File is from GameScape3D - ruined-abbey - q3-walls-bottom-a.stl (76,154KB)

GregValiant, 2022-06-28:
We just had a bug report yesterday regarding another piece of this model.
There appears to be a problem with the way that Cura if seeing some models.  Try moving it on the build plate, maybe try rotating it a little bit, or try setting your Initial Layer Height to a slightly different value (if it is 0.20mm change it to 0.21mm).  Each of those changes has been able to make a difference and allowed some models to slice.

This is a duplicate of a couple of other bug reports (#12611 is one of them and the other half of your model I think).  I will close this as a duplicate.  Thank you for the report.  You can add comments here or on the other report.

cpacker001, 2022-10-11:
I have tried that and it did not work.  So what you are telling me is that your software can not do this.  I guess it is time to find a better software.

Colt Packer