Ultimaker/Cura #12630: Negative horizontal expansion values are not displayed correctly

carribeiro, 2022-06-28:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. Loaded two models for printing (a custom pin & nut); it will print one at a time.
2. Tried to enter a negative Horizontal Expansion value.

### Actual results

The "minus" sign is not correctly rendered. It shows as an space. 

[image]

If I select the text the minus sign appears.

[image]

### Expected results

The minus sign used to work correctly. In fact I think I already did the same on Cura 5.0.0. a few days ago with no issue.

I never noticed this problem before. I'm yet to print the file and check if the results are correct (it's hard to see such small changes on the preview).

BY THE WAY -> one of the models has a few problems, the wall is too thin, I'm correcting it, but that's not something that should interfere in this case.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Project file & cura log.

[Pino com rosca integrada.zip](https://github.com/Ultimaker/Cura/files/9003396/Pino.com.rosca.integrada.zip)

carribeiro, 2022-06-28:
Note: I believe this bug could be something related to my Windows install, but then I'm not sure. It's a clean and recent Windows 10 install, with no other issues.

GregValiant, 2022-06-28:
Hello and thank you for the report.
Both models have errors.  MS 3D Builder was able to repair them.
If you try dropping the Line Width to .35 or .40 I think the print would be better.
I do not have the problem with the minus sign and it displayed correctly when I opened your project file.

carribeiro, 2022-06-30:
It's clearly a weird problem. I was able to reproduce it reliably with the following steps. Please note that opening the project file actually makes the problem DISAPPEAR - it only happens if I start with a clean bed. Check the steps.

1. Start Cura. Type -0.1 in the Horizontal Expansion text box, minus sign isn't being shown.

[image]

2. If I select the text, the minus sign is shown:

[image]

3. Then I load a 3MF with negative values, and it shows, even unselected:

[image]

4. Now it stays working after I select **File | New Project**. The bed is clear, but now the minus sign is shown.

[image]

fieldOfView, 2022-06-30:
If you look carefully at the provided screenshots, you see that there's also a difference in how the text `Horizontal Expansion` and the label above it are rendered. Something is causing both these settingsitems to be rendered at non-integer pixel offsets, which ruins the font rendering.

carribeiro, 2022-06-30:
> If you look carefully at the provided screenshots, you see that there's also a difference in how the text `Horizontal Expansion` and the label above it are rendered. Something is causing both these settingsitems to be rendered at non-integer pixel offsets, which ruins the font rendering.

Yes, I didn't notice it at first. The text seem to be "compressed". But - it's strange because the input box is a separate element in the UI. Really weird.

Joeydelarago, 2022-07-06:
This does appear to be a purely rendering issue. I agree that it is a duplicate of #12283

carribeiro, 2022-07-06:
> This does appear to be a purely rendering issue. I agree that it is a duplicate of #12283

I'd like to add that in this case, it's not just an aesthetical problem; the value initially seems to be wrong which is misleading. It took me reloading the saved project to ensure the correct value was being applied.