Ultimaker/Cura #12632: Cura Crashes when multiple monitors connected

bsspeedy45, 2022-06-28:
### Application Version

5.0.0

### Platform

OS: Windows 10, GPU: RTX3090, CPU: AMD 5950x

### Printer

Ender 5 pro

### Reproduction steps

1. When I have my 3 monitors connected search for Cura in my start menu and click on it to start.( 1 connected straight to gpu via display port, other two connected to a hub that connects to the gpu) Cura 5.0.0 will freeze/crash as soon as it begins to open. 

### Actual results

1. Cura starts loading and as it open it window it freezes/crashes.

### Expected results

1. It should just open up to an empty build plate with my printer profile like previous version.
2. But the only way I can get it to load is disconnect the hub that connect my 2 monitors ( so now I'm only using 1 monitor instead of 3) allow Cura to start and once I'm presented with the empty virtual build plate then I can reconnect connect my other 2 monitors and use Cura without issues.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Don't know if this will help but i only attempted to start Cura with only one monitor after reading a post where someone said That the issue may be happening because Cura is checking to see if a printer is connected so they suggested disconnecting all non-essential before running It. And it just happened to  work for me.(but a bit annoying disconnecting the monitors each time i want to start cura).

PS: i attempted various things prior which included
1. uninstall / reinstall (fail)
2. uninstall/ reboot /install fresh download (fail)
3. Reinstalled Windows 10 (fail)
4. checked if Cura on my laptop instead of the desktop (no issues running it).

nallath, 2022-07-05:
Duplicate of #12197

closed 2022-07-05 by nallath

PhotoJoeCA, 2024-02-24:
February 2024 Checking in using 5.4.0 and this is still an issue -- Multi-monitor system - Program hangs on launch.

KD-4-Lyfe, 2024-12-07:
still an issue 7/12/2024 cura 5.9