Ultimaker/Cura #12637: Material color request

Tundesser, 2022-06-29:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

FLSUN SR

### Reproduction steps

When setup up materials or filaments from manufacturers. I tried to enter a yellow for GTSD manufacturer. Yellow wasn't available. I was trying to make the color by duplicating a green entry.

### Actual results

Stays green.

### Expected results

should be a yellow color to choose from.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Joeydelarago, 2022-07-05:
Hello, I'm slightly confused about the issue you are having. 

You should be able to enter any hex color in the "Hex" textbox and it will use that color for your material. I've checked the hex color you have filled in with an online rgb color picker and it is the same color it shows in your dialog. 

Perhaps you are looking for some color like "#ffff00" ?

Tundesser, 2022-07-05:
Hello Joey,

The colors that popped on the screen were limited. There was no yellow shown as a choice.

I can try to enter a Hex color. I haven't had to do that before, I'm not even sure where to look them up? It seems a little rough process having to look them up rather than a simple color chart popping up to move the curser around like there was before when I used the Cura 4.13.

I'm not sure why, but I can't get the color yellow to work. The only thing I've changed is Cura 4.13 to 5.0, and then adding your tool from the "marketplace". It is strange to me also since I've never experienced this before.

Thanks,
Tom

Tundesser, 2022-10-11:
Hello Joey.

Here's a picture of my screen in Cura 5.0 that shows the color choices when I try to adjust it. I can't see any choice for yellow, and I don't know the hex code for it. Unless I'm missing something? The closest to yellow I'm seeing is orange.

Thank You,
Tom
[cid:cb26384b-61cd-463a-84c5-303579403b8f]